PowerGridModel/power-grid-model #69: [BUG] Segmentation Fault when running calculate_power_flow()

Thijss, 2022-05-12:
**Describe the bug**

I upgraded from (1.3.237 -> 1.3.254), and now I am getting segmentation faults when running `.calculate_power_flow()`

**Input Data Validity**

Did you check the validity of input data using the helper function, see examples
[here](../blob/main/examples/Validation Examples.ipynb)?

**Data Dump**
[input_for_segfault_1_3_254.json.zip](https://github.com/alliander-opensource/power-grid-model/files/8677481/input_for_segfault_1_3_254.json.zip)

**To Reproduce**

```python
input_data = load_json()
model = PowerGridModel(input_data, system_frequency=50.0)
model.calculate_power_flow(calculation_method=CalculationMethod.linear)
```

**Expected behavior**

A traceback instead of a segmentation fault would be helpful.

Thijss, 2022-05-12:
1.3.248 also still works

TonyXiang8787, 2022-05-12:
Last working version `1.3.249`
With commit hash `f30a6ca8225161478029a3479bd2f50bbe40d4ed`

closed 2022-05-12 by TonyXiang8787