PowerGridModel/power-grid-model #78: [FEATURE] Uncertainty margin for state estimation

TonyXiang8787, 2022-06-01:


jaccoheres, 2022-11-08:
@TonyXiang8787 I guess this is the a posteriori uncertainty of the estimated state?

TonyXiang8787, 2022-11-10:
@jaccoheres yes you are correct.

jaccoheres, 2024-12-09:
The use cases for this within the Alliander 'Inzicht in Basislast' team is the following:

- The state-estimator in PGM is used to kalibrate the Bottom-Up loads of PULSE per secondary substation/MV busbar to the EMS measurements at the start of every MV feeder and maybe other measurements in the feature. These kalibrated/estimated values are then used in load flow calculations, sometimes the whole loadprofile is used but more often the peak loads (sometimes within a certain time window) are used. However, due to the fact that we usually only have one measurement per MV feeder, the peak loads calculated via the state-estimation are underestimating the real peak loads, because the individual peaks are smeared out over the whole feeder. We would like to a an a posteriori uncertainty of the calculated load per timesteps, so that we could take into account the probablity that the real load is higher then the estimated load. In this way, we could estimate better peak loads.
- The estimated loads are input for forecasts, which are currently deterministic. There is however much urgence to move to probabilistic forecasts. Then the current loads should also be probabilistic in order to achieve this, so it would be helpfull to have a uncertainty of the load at all points in time.

An estimation of the uncertainty of the states could also be helpfull in deciding what the best location for new measurements would be, in order to decrease the uncertainty.

It would be of course very nice to see the influence of each parameter on the uncertainty of the estimated values, but it would be already quite something to have an idea of the uncertainty of each estimated value (voltage, current, power) on it's own, so just one positive scalar per quantity per timestep, based on the chosen sigma's.