PowerGridModel/power-grid-model #81: [FEATURE] pre-defined batch calculation

TonyXiang8787, 2022-06-01:
- N-1
- Time-series

bramstoeller, 2022-07-27:
Would this be a feature we can add to the new https://github.com/alliander-opensource/power-grid-model-io repository?

pkuppens, 2023-02-13:
@TonyXiang8787, @bramstoeller, let's hypothetically say that I'm an open-source developer, new to the power-grid-model project, and I would love to contribute by fixing bugs and implementing features.
How would I start by implementing this feature?

----
Obviously, the answer is: ask the Issuer ;)

So a few questions:

1. What Feature is exactly requested?
2. Do you have a starting point in the source code?
3. How would it be tested?
4. Do you have a pointer for testing as well? E.g. unit tests, test plans, and earlier test reports.
5. When would you consider the feature 'Done'? Is there a generic definition of done, or a specific one?