PowerGridModel/power-grid-model #91: [BUG] Batch validation for asym loads gives error

nitbharambe, 2022-06-30:
**Describe the bug**

Probably the batch validation does not work for asym loads. It must be because asym loads batch are 3d arrays while other components would give only 2d.

**To Reproduce**

```
# some basic imports
from power_grid_model import PowerGridModel, CalculationMethod, CalculationType, LoadGenType
from power_grid_model import initialize_array
from power_grid_model.validation import assert_valid_input_data, assert_valid_batch_data

# node
node = initialize_array("input", "node", 3)
node["id"] = [1, 2, 6]
node["u_rated"] = [10.5e3, 10.5e3, 10.5e3]

# line
line = initialize_array("input", "line", 3)
line["id"] = [3, 5, 8]
line["from_node"] = [1, 2, 1]
line["to_node"] = [2, 6, 6]
line["from_status"] = [1, 1, 1]
line["to_status"] = [1, 1, 1]
line["r1"] = [0.25, 0.25, 0.25]
line["x1"] = [0.2, 0.2, 0.2]
line["c1"] = [10e-6, 10e-6, 10e-6]
line["tan1"] = [0.0, 0.0, 0.0]
line["i_n"] = [1000, 1000, 1000]
line["r0"] = [0.25, 0.25, 0.25]
line["x0"] = [0.2, 0.2, 0.2]
line["c0"] = [10e-6, 10e-6, 10e-6]
line["tan0"] = [0.0, 0.0, 0.0]

# load
asym_load = initialize_array("input", "asym_load", 2)
asym_load["id"] = [4, 7]
asym_load["node"] = [2, 6]
asym_load["status"] = [1, 1]
asym_load["type"] = [LoadGenType.const_power, LoadGenType.const_power]
asym_load["p_specified"] = [[20e6,21e6,10e6], [10e6, 15e6,10e6]]
asym_load["q_specified"] = [[5e6, 5e6, 2e6], [1e6,2e6,3e6]]

# source
source = initialize_array("input", "source", 1)
source["id"] = [10]
source["node"] = [1]
source["status"] = [1]
source["u_ref"] = [1.0]

# all
input_data = {
    "node": node,
    "line": line,
    "asym_load": asym_load,
    "source": source
}

# batch data
asym_load_update = initialize_array("update", "asym_load", (2, 2))
asym_load_update["id"] = [[4,7], [4,7]]
asym_load_update["p_specified"] = [[[20e6,21e6,10e6], [10e6, 15e6,10e6]], [[31e6,31e6,20e6], [20e6, 25e6,20e6]]]
batch_data = {"asym_load": asym_load_update}

assert_valid_input_data(input_data=input_data, calculation_type=CalculationType.power_flow, symmetric=False)
assert_valid_batch_data(input_data=input_data, update_data=batch_data, calculation_type=CalculationType.power_flow, symmetric=False)

```

**Screenshots**

Error:
[image]

**Additional context**

Fix: Maybe we can add this before line 219? Would that be right/enough?
```
            if mask.ndim == 2:
                mask = mask.prod(axis=1, dtype=np.bool)
```

https://github.com/alliander-opensource/power-grid-model/blob/93a1ea54cdc23d9d68b706669ee17a7684b5a8fc/src/power_grid_model/validation/utils.py#L215-L219

bramstoeller, 2022-07-12:
I've added a unit test to highlight the bug: #2e1d70a

nitbharambe, 2022-07-14:
Thanks for adding the test. I guess the simple fix mentioned in the issue is appropriate then. Will add it.

closed 2022-07-30 by TonyXiang8787