PowerGridModel/power-grid-model #112: Resolve code smells from Sonar Cloud in Python side

TonyXiang8787, 2022-08-02:
As suggested in Sonar Cloud

https://sonarcloud.io/project/overview?id=alliander-opensource_power-grid-model

- Solve the code smell 
- Ignore it in the `sonar-project.properties` if that's the intended choice. Document in this file about the design choice.

ekiwaka, 2024-02-14:
@mgovers Sure, I will pick it up.

mgovers, 2024-07-03:
@ekiwaka how is it going? do you need any help?

VITA2aishu, 2026-09-01:
Hi, I’d like to contribute to this issue. I see it is currently unassigned and still listed as a good first issue. Could a maintainer point me to one of the current unresolved Python Sonar code smells that would be a suitable first contribution? I’m happy to prepare a focused fix and tests/checks as appropriate.

nitbharambe, 2026-09-01:
Hi @VITA2aishu , you can find them here: https://sonarcloud.io/project/issues?languages=py&s=IMPACT_RANK&issueStatuses=OPEN%2CCONFIRMED&id=PowerGridModel_power-grid-model.