PowerGridModel/power-grid-model #132: [FEATURE] Tuning tap changer position of distribution transformers in state-estimator

jaccoheres, 2022-09-15:
From Insight into Loads, we would like to use the voltage measurements from FlexOVL smart meters that usually are situated on/near the LV rack of a secondary substation, and include these in a state-estimation calculation. A challenge here is that the step position of the distribution transformer can have a significant influence on the effect of this measurement on the estimated state of the rest of the (MV) grid.  I was wondering if you see possibilities for including the uncertainty in the tap position of the transformer in the state-estimation of the power-grid model. So e.g. when the tap position is set to 3 but with position 4 the total residuals of the measurements would be significantly lower, choose that position. My first suspicion is that this isn't very trivial. It seems easier to determine this over several time steps, if the load and thus the voltage drop is very low, a wrong step position would be easier to detect than with a high load. But then a calculation/analysis over several time steps would be necessary. But if this feature works well, it can also mean a nice data improvement that yields much more than just for our team.

jaccoheres, 2022-11-02:
@TonyXiang8787 yes that is correct

jaccoheres, 2022-11-09:
@TonyXiang8787 I was thinking about this for a while, and it might maybe be easier if the tap position is not part of the unknowns but that a analysis is done on the state-estimation results to find probably wrong tap positions. In this way it is easier to combine the results from multiple timesteps.

AnkurArohi, 2023-06-08:
Sorry that I jump here in the middle of discussions, I have a small suggestion to male -> please do not include tap position into the state estimator, rather I suggest the following

- Run WLS as before estimation voltage and the angles
- After the run based on valid and calculated constraints try to determine the quasi state change based on the changes in the tap positions (This module is referred by us as Post processor)
- In the post processor module we try to change the  tap positions considering the valid constraints and validate the results with A. Valid low error measurements if available and B) Load flow calculation (also for the microogrid/island network)
- After the post processing we need a validator which validates the results and provides the initial state, estimated state, quasi estimated / post processed state 

In this way one has deep knowledge of the states and can understand what are the possible tap positions.

Welthulk, 2024-01-16:
Yes, that could be an approach. However, I'm also aware of the possibility of including the additional voltage of the transformers in the estimation matrix. By using the estimated additional voltage, it's possible to recalculate to the transformer taps. But, my concern was to inquire whether your state estimation supports transformer tap estimation.