PowerGridModel/power-grid-model #140: [FEATURE] Alert user to unrealistic load value

hansrosenberg, 2022-10-06:
We experienced a failure to converge using the model. Analysis showed later that this was due to an extremely high power value that was assigned to a load.

Maybe it is possible to flag unrealistic load values or the highest load value in the topology.

One other possible way is to make the max load threshold depend on the voltage level. For instance, 10MW at a node that carries 400V is not very realistic. Maybe a realistic max value for that would be something like 500kW so setting a limit at 1MW is probably a nice value. For 10kV more than 10MW on a single load point may start to become unrealistic (allthough the spread in loads there can be much higher). You could flag these kinds of values as 'suspicious' and leave it to the user to decide weather this is the case, assuming the user knows the local situation and can determine how realistic a value is.

hansrosenberg, 2022-10-06:
I suggest to add this to the validate_input_data() function

jaccoheres, 2022-11-02:
Would it be possible to better pinpoint the reason of a failure to convergence (that could be because of an unrealistic high load)? This is of course not possible in validate_input_data() but only after applyting a loadflow or state-estimation.

hansrosenberg, 2022-11-28:
Hi Jacco, we had a short brainstorm with the dnr team a few weeks ago since I asked this exact same question. I think the outcome was that it should be possible to make a 'validate loads' function or something like that that would check some power limits for each voltage. For instance, for 400V, you can set some limits for source/load values outside of which values are unrealistic/unlikely and present those to the user as a warning. A warning, since you cannot be 100% sure that it is really not true, maybe you have a bit of an outlier case. You would specifically have to enable this validate function (just like the other validate functions in the pgm package).

Best regards,

Hans