PowerGridModel/power-grid-model #169: [FEATURE] Load/Generation component with voltage characteristic

nitbharambe, 2022-12-21:
A load/generation component with a manually provided voltage-power characteristic as input is needed for modelling a Photovoltaic or a motor.

GSkrt, 2023-05-06:
This is actually very important feature this days. Reactive power of converters currently inside most of the DG has huge impact on voltage and thus on hosting capacity of each network bus. Having this evaluations within a main loop for solving LF can have huge impact on solution speed and number of necessary iterations.

mgovers, 2023-06-02:
i think this is a duplicate of #49 (or the other way around)

TonyXiang8787, 2023-06-02:
@mgovers they are completely different.

GSkrt, 2023-09-17:
In mean time I can do double iterative process (loadflow 1 ),  loadflow->U (DG node) -> set new Q in a loop till there is little or no difference between iterations on node voltages.  It would be faster if it used same iterative process as LF ? Its almost the same problem as setting tap changers on transformers,  just that it uses linear function instead of discrete one.

TonyXiang8787, 2023-09-18:
Hi @GSkrt, just to be sure what you mean here. Mathematically, we are talking about a voltage dependent load, correct? So the active and reactive power of a load is a piecewise linear function of voltage magnitude.

$$P = P_0 f_P(V)$$

$$Q = Q_0 f_Q(V)$$

We are actually planning to directly include these characteristics in the Power Flow solver. So we do not need double iterations. Same as for automatic tap changers, we plan to do this in one go if possible.

The main complication of this one is not on the solver or the mathematical model, but on the data models. Characteristics are variable length input. We need define this in the data interface.

GSkrt, 2023-09-24:
**"Mathematically, we are talking about a voltage dependent load/generation, correct?"** 

That's correct. When defining f, you should consider that this functions are not only linear but can be any function. 

Reactive power has a big impact on voltage when calculating impact of DG-s on voltage profile in MV network that has low R/X ratio. 

It could be also a combination of discrete and linear function and that makes implementation of such solver a bit of a problem. If I want to generalize the transfer function I could say: 

$$ S_v = \text{any system state variable/s } $$

Where system state variables could be any model result variable not only V at connection node (I,V,P,Q, switch state  or frequency deviation of the system) but also other variables such as state main transformer tap changer. Reason for that is that there could be very complex voltage regulation strategies in place, that don't consider only voltage at connection node.

$$ Q_{c1} =  \mathbb{C} \Rightarrow x_0 < S_v < x_1 $$

$$ Q = Q_0 f(S_v) \Rightarrow  x_1 < S_v < x_2 $$

$$ Q_{c2} =  \mathbb{C} \Rightarrow x_2 < S_v < x_3 $$

Below is a common reactive power curve that can be enabled on most PV inverters: 

[image]

The P(V) is not used very common except P(f) and P(system state) for flexibility reasons. 

All above is not easy to implement, but it would be very useful tool to have to optimize network operation strategies.

mgovers, 2023-10-02:
> So the active and reactive power of a load is a piecewise linear function of voltage magnitude.

One maybe very irrelevant side remark is that I don't know whether piecewise linear is sufficient. Shouldn't the function be $$C^1$$?

TonyXiang8787, 2023-10-04:
Hi @GSkrt thanks for you input. I think we are going to start with voltage-dependent characteristic first. Using other state variable to control P/Q will be more challenging and we need to think about that.

Do you think if voltage-dependent characteristic alone can already be beneficial for your use-case?

GSkrt, 2023-10-28:
Well if we wanted to simulate reactive power from sinchroneous generators cos/sin should be implemented (due to reactive power limits that are not linear) but I guess most of DGs this day have inverters with some reactive power limits, so piecewise linear is indeed enough for most of the cases for DSO.

TonyXiang8787, 2023-10-28:
> Well if we wanted to simulate reactive power from sinchroneous generators cos/sin should be implemented (due to reactive power limits that are not linear) but I guess most of DGs this day have [...]

@GSkrt I think if we give enough points to the piece wise linear function, it can still represent sin/cos reasonably.

GSkrt, 2023-10-28:
Ok that could work.