PowerGridModel/power-grid-model #185: PV (voltage controlled generator) as slack bus

TonyXiang8787, 2023-01-09:


TonyXiang8787, 2024-08-10:
@sudo-ac yes, this issue is meant to define PV nodes. However, PV nodes rarely (to none) exist in distribution grids. This project is becoming more fine-tuned for DSOs. So this issue is likely to be closed.

TonyXiang8787, 2024-08-14:
> Do you mean that you will never implement PV nodes in PGM?

Unless there is a major shift of the focus of PGM, it is unlikely to implement PV nodes.

petersalemink95, 2024-08-26:
@sudo-ac thank you for the interest in this issue. Since the focus of the Power Grid Model project is mainly on distribution systems it is unlikely that PV nodes will be included in Power Grid Model, unless there comes a shift in focus. Therefore I am closing this issue.

petersalemink95, 2025-08-26:
@scud-soptim That would be great! Let us know if you do - we'd be happy to think along!