PowerGridModel/power-grid-model #186: Optimal Power Flow

TonyXiang8787, 2023-01-09:


Flinverdaasdonk, 2023-05-11:
Would be interested! Beneficial for my research to determine which control options are most effective to reach certain goals.

kornerc, 2026-05-04:
> Important thing to add (please add a 1-line edit to the original reply to reflect this to make sure that other people reading the comment are not confused): This Julia-based approach (or any 3rd-party implementation) is for a first experiment only. The final solution that can be accepted into PGM/PGM-DS should be in either Python or C/C++. The PGM and PGM-DS are lightweight packages with relatively few dependencies. We can't demand that users install a Julia runtime just to install PGM in a Python environment.

Thanks for the heads up; I've added such a disclaimer.