PowerGridModel/power-grid-model #189: Expose internal data (Jacobian, Node admittance matrix)

TonyXiang8787, 2023-01-09:


Jerry-Jinfeng-Guo, 2024-11-07:
In the research community, access to internal parameters (per iteration) can be of great help for, e.g., training purposes.

bartgips, 2024-12-17:
Exposing the internal data could be useful when someone is building a machine learning model with PGM in the loop.
e.g. one could train a neural network to perform state estimation, see e.g. : https://ieeexplore.ieee.org/abstract/document/8851855

If PGM could somehow give the user a gradient of the likelihood/loss function for all state variables (V, phi) this could be propagated to parameters of some machine learning model.

As Jerry notices above, this would require PGM to be used as a "derivative calculator" only. I.e. only a single iteration is enough (no full optimisation run). This means it would need to take a state (V, phi) as input (next to batch data that contains measurements), and would give gradients (and perhaps Hessians) as output.

bartgips, 2024-12-17:
Besides this, it would perhaps be helpful if internal values such as the node admittance matrix would be accessible to the user. In our case it would have been useful to double check if our PyTorch code is faithfully following all PGM conventions.
Besides this, it could be useful for didactical reasons (e.g. teaching students about state estimation)

behzadbehdani, 2026-08-25:
I am currently working within DEP on a project at Alliander for SO and TCM, analyzing the sensitivity of system voltages to variations in active power. The objective is to predict voltages at specific system nodes and determine the amount of active power required to regulate those voltages.
To perform this analysis, I am using PGM. Access to the Jacobian matrix, in particular, as well as the nodal admittance matrix, is required. Therefore, exposing the nodal admittance matrix and the Jacobian matrix from the final iteration of the power flow calculation would be necessary to carry out these analyses.