PowerGridModel/power-grid-model #193: [BUG] Program will crash if you have NaN in the input, but valid data in batch set

TonyXiang8787, 2023-01-18:


mgovers, 2023-04-26:
@TonyXiang8787 isn't this enough to just fix everything?

```cpp
        // calculate once to cache topology, ignore results, all math solvers are initialized
        (this->*calculation_fn)(std::numeric_limits<double>::infinity(), 0, calculation_method);
```

mgovers, 2023-04-26:
> @TonyXiang8787 isn't this enough to just fix everything?
> 
> ```c++
>         // calculate once to cache topology, ignore results, all math solvers are initialized
>         (this->*calculation_fn)(std::numeric_limits<double>::infinity(), 0, calculation_method);
> ```

After discussion: this does not suffice. There are direct calculation methods that do not use tolerance and iterations

PR #250 (2023-05-02): Bugfix/incomplete input valid update
* Allow `infinite` error tolerance to be passed to the math solver.
  * Accept any valid output
  * Prevent fatal crashes when solution is ambiguous (e.g. not invertible)
* Small refactor of math solver for readability
  * Use `switch`/`case` statement to select solver
  * Remove hardcoded values for `linear_current` solver
* Tests that show this solves #193

Closes #193
closed 2023-05-02 by TonyXiang8787