PowerGridModel/power-grid-model #237: [FEATURE] Possibility to specify different sigma's for P and Q in state estimator

marionsnijders, 2023-04-12:
Within team Insight into Loads we apply the state estimator to the MS grid. We see an issue in our output related to the restriction (if we understand correctly that this is the case) of one sigma for both the P and Q of a (pseudo-)measurement. We have the following inputs:

- MSRs (loads) containing lots of households: those will have a larger margin of insecurity for P due to sampling of anonymous user profiles.
- MSRs containing just one large consumer (AC5): those will have a very small margin of insecurity for P due to an available measurement
- Measurement of P and Q at the substation (afgaande velden): both P and Q have a small margin of insecurity here.

In general, we expect a low Q for households and a possibly higher Q for the larger customers. Unfortunately, because we specify one sigma for P and Q, the reactive power is largely attributed to the households now whenever the measurement at the substation indicates that there is reactive power on the route. This makes the output of such routes/grids invalid for our application. Would it be possible to specify two different sigma's within one (pseudo-)measurement within the state estimator, so we can start influencing the attribution of Q more accurate?

TonyXiang8787, 2023-05-11:
Decisions:

1. Expand the current iterative method to decouple the real and imaginary part. So it can handle different sigmas.
2. In the power sensor, add a new optional attribute `reactive_power_sigma`. If it is specified, `power_sigma` becomes only for active power. Otherwise, `power_sigma`  is for complex power.

mgovers, 2023-11-29:
Update 2023/11/29:

* Since #409 , the user may already use `sigma_p` and `sigma_q` to override `power_sigma`.
  * For certain systems, this will already provide some improvement over using `power_sigma` by itself.
  * See https://power-grid-model.readthedocs.io/en/stable/user_manual/components.html#id28 on details, input requirements, ... .
  * See https://power-grid-model.readthedocs.io/en/stable/user_manual/calculations.html#state-estimation-algorithms on details on the treatment for the implemented calculation type(s)
* Implementation work on #369 has started to add the Newton-Raphson calculation method for state estimation
  * This should unlock the full potential of separating `sigma_p` and `sigma_q`

mgovers, 2024-02-08:
Update 2024/02/08:

#369 is now in beta. For details, please have a look at https://github.com/PowerGridModel/power-grid-model/issues/369#issuecomment-1934542251

PR #499 (2024-02-16): Release v1.7.0: Newton-Raphson State Estimation
Closes #369
Closes #237 

* Bump version
* Make Newton-Raphson state estimation stable (instead of experimental)

* [x] full clang-tidy run in https://github.com/PowerGridModel/power-grid-model/actions/runs/7927335575

Re-creation of #497 and #498 due to `release` in branch name
closed 2024-02-16 by TonyXiang8787