PowerGridModel/power-grid-model #254: [FEATURE] threading for multiple models, instead of one model with multiple update_data

leiwu84, 2023-05-10:
My use case usually requires say 100 models, i.e. 100 instances of `PowerGridModel`. At this moment, its batch calculation feature only allows one model with many different update_data. I would suggest allowing batch calculation for many models. Then I can use multiple threads to run power flow for all the models in parallel. For example, I can run all 100 power-flows in one round if I have 100 threads, or five rounds with 20 threads, and so on.

Jerry-Jinfeng-Guo, 2024-06-11:
https://github.com/PowerGridModel/power-grid-model/issues/128

mgovers, 2024-06-11:
> #128

@Jerry-Jinfeng-Guo should we close this ticket as duplicate of #128 (or vice versa)?

Jerry-Jinfeng-Guo, 2024-06-11:
> > #128
> 
> @Jerry-Jinfeng-Guo should we close this ticket as duplicate of #128 (or vice versa) (there's a label for that)?

Probably a good idea. @leiwu84 We will combine this feature request with #128. Please find related updates there.

closed 2024-06-11 by Jerry-Jinfeng-Guo