PowerGridModel/power-grid-model #259: [FEATURE] *Provide the power system mismatch*

SanPen, 2023-05-25:
It would be nice if the results would provide the nodal power mismatch or the norm of this.

This is the stopping criteria for Newton Raphson.

For methods that use voltage iteration similarity as criteria (i.e. Backwards-Forward) provide the power mismatch as well.

TonyXiang8787, 2024-09-20:
The power flow results already include the node injection power which is calculated by network parameters. At the same time all appliances have their power values by appliance parameters.

The difference of these two is the power mismatch. I am hesitating to provide additional attributes for the mismatch. The user can easily calculate this by himself. @petersalemink95 what do you think?

SanPen, 2024-09-30:
Hi,

Maybe not the mismatch vector but the norm `max(abs(mismatch))` since this is a useful value to decide if to trust the solution.

mgovers, 2024-09-30:
a potentially useful consideration here is the following:

* for development/desktop/hobby project usages:
  * memory is generally not the bottleneck
  * the computational overhead of adding such a feature is negligible
  * adding an attribute like this should therefore be alright
* for production environments at scale:
  * memory may be a real bottleneck.
  * for row-based data structures (the main data structure used in the PGM):
    * adding such an attribute impacts memory 
    * the addition of a parameter like this is therefore discouraged (the main argument used in the above)
  * however, we are currently working on columnar data support (#548)
    * this feature is already experimental and will be the main new feature in the release associated with next minor version bump.
    * production environments almost always work with columnar databases
    * because of that, they are likely to use this new feature.
    * in particular, the feature allows omitting attributes from the output that are not needed for the use case.
    * as a result, the argument used before may not affect production environments too much in the future.

TonyXiang8787, 2024-09-30:
@mgovers even with columnar dataset we are not dealing with memory issue. The calculation itself still cost some CPU time. It might not be very much.

nitbharambe, 2024-10-09:
Detailing the steps for finding mismatch in power after calculation. Correct me if I am wrong:

1. Calculate power flow
2. Scale the specified powers of loads/gens in input_data of `const_impedance` and `const_current` type by result's voltage.
3. Aggregate all loads/gens of the input_data to node level by summing them up
4. Subtract with result's nodal power injection and find maximum among them.

How about we provide a utility function to do this? 
Or what if we have this as addition for the PGM data science toolkit? 
This problem is a post calculation operation on numpy arrays. Similar might be applicable to bad data detection with chi square test feature request.