PowerGridModel/power-grid-model #263: [FEATURE] Way to extend components with custom fields

Flinverdaasdonk, 2023-06-06:
To extend PGM functionality, I think it would be beneficial to have a simple way to add custom fields (e.g. "id", "from_node", etc.) to PGM components (nodes, lines, etc.). 

For example, currently I'm implementing ACOPF. ACOPF is a way to determine (among others) optimal set points for generators. I intend for the implementation to be easily compatible with PGM. For ACOPF I need additional fields per component, e.g. for a load the minimum and maximum load. Currently, it is not possible to easily add these fields, since a numpy structured array does not allow this.

Examples of how the functionality could work are
- python dictionary style: sym_load["max_power_consumption"] = np.array([100, 200, 300], dtype=...)
- Making it easy to make new components using inheritance, and adding the fields there

I think this feature would be valuable for other people working on extending PGM.

TonyXiang8787, 2023-06-07:
Hi @Flinverdaasdonk  Thanks for you suggestions. Indeed with numpy structured arrays it is difficult to add new attribute. On the other hand, this is extremely efficient to interact with the C++ core. So this is a design choice.

That being said, we have a proposal to build another abstraction layer above the numpy structured array, which should support IDE auto completion and allow users to extend the array. Internally, all the user defined attributes will be stored in separate dictionaries of arrays. So it will not impact the calculation core.

This issue is related to #82.

mgovers, 2024-02-14:
I made this ticket's expertise `API and Design` for the reason that it needs to be refined more first, before implementation in Python may be started

petersalemink95, 2025-12-16:
This is possible with Power Grid Model DS. For an example see: https://power-grid-model-ds.readthedocs.io/en/stable/examples/model/grid_extensions_examples.html

Since this is possible in PGM-DS, I'm closing the issue