PowerGridModel/power-grid-model #365: [IMPROVEMENT] Input/output/update data structures as standard layout

mgovers, 2023-08-29:
## Background

The underlying [input, output and update data structures](https://power-grid-model.readthedocs.io/en/stable/user_manual/components.html) behind the data types retrieved by [initialize_array](https://power-grid-model.readthedocs.io/en/stable/api_reference/python-api-reference.html#power_grid_model.initialize_array) are generated using [code generation](https://github.com/PowerGridModel/power-grid-model/tree/main/code_generation).

The current implementation uses inheritance to define the relationship between components and the types they implement and are derived from. This keeps the data structures short and readable.

## Problem

The downside is that inheritance causes the data types are not guaranteed to obey standard layout. This may result in less efficient data structures, and also makes the code more bug-prone. As an example, #241 was caused by the fact that the data type does not have standard layout.

## Solution

The standard layout solution can be achieved using a plain struct types without references to any base classes. For written code, that would cause a lot of code duplication, but since the code under consideration is generated, this is not an issue.

The notion of the "base" types are enabled by declaring them `concept`s, which can be tested using `static_assert`.

```cpp
template <typename T>
concept base_input = requires (T t) {
    { t.id } -> std::same_as<ID>;
};

template <typename T>
concept node_input = base_input <T> && requires (T t) {
    { t.u_rated} -> std::same_as<double>;
};

template <bool sym>
struct NodeInput {
    ID id{};
    double u_rated{};
};

static_assert(base_input<NodeInput<true>>);
static_assert(node_input<NodeInput<true>>);
static_assert(base_input<NodeInput<false>>);
static_assert(node_input<NodeInput<false>>);
```

Constructors of components will then take the following form (and equivalent for their update and output functions):

```cpp
class Base {
  public:
    explicit Base(base_input auto const& input_data): id_{input_data.id} {}

  private:
    ID id_{};
};

class Node: Base {
  public:
    explicit Node(node_input auto const& input_data): Base(input_data), u_rated_{input_data.u_rated} {}

  private:
    double u_rated_{};
};
```

This also will make the initializer list constructors a lot more readable for e.g. testing purposes.

## Other considered options

Another option could be to use [composition](https://en.wikipedia.org/wiki/Composition_over_inheritance), e.g. using data types like

```cpp
struct BaseOutput{
  ID id{};
};
struct DerivedOutput{
  BaseOutput base{};
  int other_field{};
};
```

but it becomes prohibitively non-trivial to query data fields of underlying types without getters and setters (e.g. we would need `transformer_output.line.base.id` to query the `id` field of a transformer).

TonyXiang8787, 2023-08-29:
@mgovers To avoid confusion, I would propose to use a fixed suffix for concepts defined in the input/output data. For example `*_c`, `base_input_c`.

mgovers, 2023-09-11:
@TonyXiang8787 my intellisese works on the concepts solution. clang-tidy and code sonar running on the other in #376

mgovers, 2023-09-12:
@TonyXiang8787

* For the implicit reference option
  * Sonar Cloud wants to add the `explicit` keyword, which requires adding an explicit `static_cast` at the usage.
  * Sonar Cloud wants to replace the `reinterpret_cast` which we would have to disable explicitly
* For the proposed concepts solution:
  * No issues in clang-tidy and Sonar Cloud
  * Intellisense provides member suggestions correctly

TonyXiang8787, 2023-09-12:
> @TonyXiang8787
>
> * For the implicit reference option
>
>   * Sonar Cloud wants to add the `explicit` keyword, which requires adding an explicit `static_cast` at the usage.
>   * Sonar Cloud wants to [...]

For implicit cast, we can disable both implicit cast and reinterpret cast warning from SonarCloud. 

For concept approach, good that the IDE recognizes.

Please put both options into candidate approaches, we will decide when we are actually doing this.

PR #430 (2023-11-22): Feature/standard layout input
Fixes #365 

Implements all input, output and update data structures as standard layout by using flat attribute expansion in the generated code. conversions to former base types is now done using `reinterpret_cast`

Relates to #241 and #276
closed 2023-11-22 by TonyXiang8787