PowerGridModel/power-grid-model #431: Remove the Dataset logic from PGM core, use DatasetHandler for MainModel

TonyXiang8787, 2023-11-21:
# Background

Currently we use [`Dataset`](https://github.com/PowerGridModel/power-grid-model/blob/95212803cd9af486066deeea4d334b558ccf6e1a/power_grid_model_c/power_grid_model/include/power_grid_model/auxiliary/dataset.hpp) as the input view for `MainModel`, such as:

https://github.com/PowerGridModel/power-grid-model/blob/95212803cd9af486066deeea4d334b558ccf6e1a/power_grid_model_c/power_grid_model/include/power_grid_model/main_model.hpp#L101C6-L101C6

The C-API receives `DatasetHandler`, we have a function to export to a `Dataset`. This conversion is redendant. We already have `DatasetHandler` which handles datasets.

mgovers, 2024-02-14:
@TonyXiang8787 how much of this is still relevant / unblocked with the new `constexpr` metadata from #475 ?

TonyXiang8787, 2024-02-19:
@mgovers still quite relevant

PR #587 (2024-05-31): Feature/remove legacy dataset
Closes #431 

Also make `meta_data` as input argument for many classes, instead of hard code to the automatically generated `constexpr` version. In this way, it is possible to do unit test with mocked `meta_data`.

# Check list

- [x] Calculation core
- [x] C API
- [x] C Example
- [x] C API test
- [x] Unit test
- [x] Integration test
- [x] Validation test
- [x] Benchmark
closed 2024-05-31 by TonyXiang8787