PowerGridModel/power-grid-model #435: [FEATURE] Compile-time meta attribute offsets

mgovers, 2023-11-21:
**Describe the feature request**
Add the offsets for each component attribute to the [generated metadata attribute setters](https://github.com/PowerGridModel/power-grid-model/tree/main/power_grid_model_c/power_grid_model/include/power_grid_model/auxiliary/meta_gen) using [`offsetof`](https://en.cppreference.com/w/cpp/types/offsetof) (defined in `<cstddef>` in the STL) to replace the run-time determination using [`get_offset`](https://github.com/PowerGridModel/power-grid-model/blob/main/power_grid_model_c/power_grid_model/include/power_grid_model/auxiliary/meta_data.hpp#L20-L28).

However, that is a C Macro - not a template, meaning that the attribute needs to be obtained directly from the attribute name - not as a template. The template getters used later in the code should be generated by the code generator for each meta attribute.

**Background**
In #276 , the way metadata attribute offsets were calculated was modified to account for the fact that derived objects are not guaranteed to be `std::standard_layout_v`(see in particular #241 and [this change](https://github.com/PowerGridModel/power-grid-model/pull/276/files#diff-2bac4fa15db710784f8450b130e98864e112b2b5e4b104c0470f876d5894f4eeL138-R141)). In #430 , the generated component meta data structures were changed to plain structures, removing the need for runtime `inline` offset getters.

TonyXiang8787, 2023-11-22:
@mgovers can we go a step further. To define all the meta data in a fully static/global `constexpr` struct? The main challenge is the variable length of the attributes and components. I guess we can define a `static constexpr std::array<MetaAttribute, N>` and `constexpr MetaAttribute const* data; constexpr size_t n_attributes`.

mgovers, 2023-11-22:
> @mgovers can we go a step further? To define all the meta data in a fully static/global `constexpr` struct? The main challenge is the variable length of the attributes and components. I guess we can define a `static constexpr std::array<MetaAttribute, N>` and `constexpr MetaAttribute const* ptr_attributes; constexpr size_t n_attributes;`.

i think we can because the registering of component attributes happens in generated code. If we there also register the component type and offset, we can get rid of most of the complexity in the `MetaAttributeImpl` and `trait_pointer_to_member`. I don't foresee a good design for that at the moment though...

TonyXiang8787, 2024-01-20:
Ways of using variable length array:

https://godbolt.org/z/bT7nfEr17

PR #475 (2024-01-22): Use constexpr for all meta data
# Intro

This should fix #435. All meta data are now known at compile time. The whole meta data is a `constexpr` struct.

# Approach

Because the variable length of components and attributes. We define some meta-data getter struct which contains `static constexpr std::array` of attributes and components. Then we expose this array into the real meta-data in a `std::span`.

# Todo

- [x] Remove the `meta_data` function. Make it directly as a `constexpr` global member.
- [x] Fix the checks.
closed 2024-01-22 by mgovers