PowerGridModel/power-grid-model #465: Add type hinting to Python API

mgovers, 2024-01-11:
See the TODO in https://github.com/PowerGridModel/power-grid-model/blob/81b1f8c900152fd32ce64e86cb3fb0edb4f2b7d7/src/power_grid_model/data_types.py#L13

Support for Python 3.8 was dropped in #464

Cbit11, 2024-01-11:
I'll do it

mgovers, 2024-01-11:
Awesome! LMK if you need any help

mgovers, 2024-01-30:
@Cbit11 how is it going with the development?

Jerry-Jinfeng-Guo, 2024-02-15:
I don't understand this issue yet, what's wrong with this [data_types.py](https://github.com/PowerGridModel/power-grid-model/blob/81b1f8c900152fd32ce64e86cb3fb0edb4f2b7d7/src/power_grid_model/data_types.py#L13) document?

mgovers, 2024-07-04:
> this may already be done in #613 and #648

this is actually not entirely true.

* Starting python3.9, we have `list[Foo]` and `set[Foo]` instead of `List[Foo]` and `Set[foo]`, as well as `collections.abc.Iterable` and `collections.abc.Mapping` as opposed to `typing.Iterable` and `typing.Mapping`.

* Starting python3.10 ( #638 ) , we will have support for `TypeVar` including `bounds` https://docs.python.org/3.10/library/typing.html#typing.TypeVar . This will enable us to remove syntax like `Union[SingleDataset, Dict[str, SingleComponentData]]` and replace it with `SingleDataset`, where `SingleDataset = Dict[ComponentTypeVar, SingleComponentData]` and `ComponentTypeVar = TypeVar['ComponentTypeVar', bounds=[ComponentType, str]]`
* Starting python3.10, `Union` is deprecated and replaced with the much simpler syntax `|`

PR #685 (2024-08-15): Implementation of TypeVar in more scripts
Improve type hinting that is present to Python 3.10 standards (e.g. replace Union with |)
Add type hinting to places where it’s not present yet (make use of TypeVar)

* part 2 of #465

Closes #465
closed 2024-08-15 by mgovers