PowerGridModel/power-grid-model #466: [BUG] ValueErrors on MacBook on >=v1.6.22

mgovers, 2024-01-11:
**Describe the bug**

A clear and concise description of what the bug is.

**Input Data Validity**

Did you check the validity of input data using the helper function? 
[Examples here](https://power-grid-model.readthedocs.io/en/stable/examples/Validation%20Examples.html)

**Data Dump**

Please dump the input (and maybe also the output) data into json files using the helper function, if applicable
[Examples here](https://power-grid-model.readthedocs.io/en/stable/examples/Make%20Test%20Dataset.html#export-to-json)

**To Reproduce**

Steps to reproduce the behavior, including some calling code.

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

TonyXiang8787, 2024-01-11:
@mgovers there is no description of the bug?

mgovers, 2024-01-11:
Will be added to in the near future. I asked the reporter to add to this issue

mgovers, 2024-01-19:
unit tests + validation tests pass on Jean's MacBook. Working on a minimal repro case

closed 2024-01-24 by TonyXiang8787

mgovers, 2024-01-24:
Both independent reporters @Thijss and Jean confirmed the issue to be resolved by #477 (released in `v1.6.55`).