PowerGridModel/power-grid-model #486: [FEATURE] Optimize compilation speed by reorganize headers

TonyXiang8787, 2024-02-05:
# Introduction

The calculation core is a header-only library in C++. In each build target of unit test, C API, validation test, etc, we have multiple translation units (`.cpp` files). The compilation speed is slow in both CI and local development machine. This is frustrating and slows down the development.

# Proposal

We would like to re-organize the headers in the header only library to reduce the dependencies between the headers. Hopefully we can increase the compilation speed. Possible ways of improvement can be:

1. Remove most of the standard library include in the main file [`power_grid_model.hpp`](https://github.com/PowerGridModel/power-grid-model/blob/main/power_grid_model_c/power_grid_model/include/power_grid_model/power_grid_model.hpp). Only keep the absolute common standard libraries (like `cstddef`). Include the needed standard library in individual header files where it is needed.
2. Forward declare classes in the main file [`power_grid_model.hpp`](https://github.com/PowerGridModel/power-grid-model/blob/main/power_grid_model_c/power_grid_model/include/power_grid_model/power_grid_model.hpp), so that other headers can use the opaque type if the full type is not needed. This reduces the dependencies between the headers.
3. Use separate header file with `type_traits` types to specify the `input`, `update`, .. types of each component. In this way, when compiling meta-data (like serializer), we do not need to include all the component types, only the traiter types.

mgovers, 2024-02-05:
I also think that the fact that we do not use builder patterns increases this issue. that's why e.g. test_math_solver.cpp needs **all** solvers, even though not all solvers are needed for all tests

We can do something with the math solvers in the same way as we do for the `AllComponents` in the `MainModelImpl`.

I also would recommend putting the `using` declaration in a separate header file, such that the tests that do not care about e.g. the iterative linear solver does not need to know about that solver.

Making smaller test files should also help, because `doctest` spends a lot of compilation time just on uncollapsing e.g. nested subcases

mgovers, 2024-02-05:
irrelevant for CI but relevant for local development: i also think that we can split up the build target of the unit tests into multiple smaller ones to prevent the need of compiling all tests even though you're only looking into the deepest one

TonyXiang8787, 2024-02-05:
@mgovers please modify the main content with your suggested header reorganization.

TonyXiang8787, 2024-02-05:
> irrelevant for CI but relevant for local development: i also think that we can split up the build target of the unit tests into multiple smaller ones to prevent the need of compiling all tests even though you're only looking into the deepest one

This could be the next step if the compilation speed is still not satisfying after optimizing the header structure. My guess is that the target compilation speed should be improved (with incremental build) when the headers are decoupled a bit.

Jerry-Jinfeng-Guo, 2024-02-21:
This might be very novice input, but aren't we making heavy use of templates? I think normally that is also not the easiest thing when it comes to compiling.

Jerry-Jinfeng-Guo, 2024-02-21:
I tried compiling this branch (e.g.) and it seems the test cases are taking up a significant portion of the total compilation time. 
[image]

TonyXiang8787, 2024-09-20:
@mgovers 

Is this issue still relevant? After last discussion we might choose a different way of approach?

mgovers, 2024-09-24:
Although this ticket has majorly been repurposed compared to the initial setup, I think @figueroa1395 's work on getting the #718 should close this ticket.

PR #718 (2024-10-07): Feature/Re route validation tests
Re-route the validation tests through the C API:
- [x] Move the call for the tests to the C++ wrapper of the C API.
- [x] Remove the PGM core dependencies.

~~Re-route the benchmarks through the C API:~~
~~- [ ] Move the call for the benchmark to the C++ wrapper of the C API.~~
~~- [ ] Remove the PGM core dependencies.~~

Note: The re-routing of the benchmark won't be addressed here anymore, since it requieres more effort than initially anticipated. For the moment, it is taken out of the CI build and addressed on the nightly build #736 

Closes #486
closed 2024-10-07 by mgovers