PowerGridModel/power-grid-model #493: [FEATURE] Add custom attributes for serialization

TonyXiang8787, 2024-02-13:
# Background

Currently the (de-)serialization only supports attributes which are defined by PGM self. Users wish to add custom attributes to the (de-)serialization format and parser. Following attribute types can be 

- Attributes per element
- Aggregated attributes along all elements for a given type and scenario
- Aggregated attributes along all scenarios for a given type and element
- Global attributes for a given type

# Approach

To be continued

IrenaDk, 2024-02-13:
This would be very useful for our product because we do a lot of manipulations on the initial PGM network that we use and it would be useful to (de)serialize it with the additional attributes. Such output can be used as input in the next steps of our workflow, final output, and has the additional benefit of making debugging easier.

mgovers, 2024-02-14:
Initial thoughts: we could go a number of ways:

1. always aggregate. the results would end up in the meta data of the output. this marks a compatible change tot the serialization schema. the difficult decision here is whether it should be ignored or checked for `strict=true` when deserializing
2. add optional flags to aggregate across multiple dimensions and output.
3. similarly to the `extra_info` attribute to the PgmJsonConverter: make this optional and output "all the rest" as an new string of the same data type as the input. This would not break current functionality and with the visitor pattern, it should be relatively straight forward to produce, but i do not think this enables all of the above requested features