PowerGridModel/power-grid-model #500: [FEATURE] PRs possible from branches that contain `release`

mgovers, 2024-02-16:
**Describe the feature request**

Currently, it's not supported to make PRs from a branch that contain the string `release`. Not only `release/**` but also `feature/release-*` is not supported. See also #497 and #498.

TonyXiang8787, 2024-02-16:
My proposal is to remove the `release` as a whole. That means:

1. Trigger `push` event in workflow only on `main`.
2. Only upload wheels and make release for `main`.
3. Remove the `rc` versions in the release strategy.

mgovers, 2024-02-16:
> My proposal is to remove the `release` logic as a whole. That means:
> 
>     1. Trigger `push` event in workflow only on `main`.
> 
>     2. Only upload wheels and make release for `main`.
> 
>     3. Remove the `rc` versions in the release strategy.

i'm very much for this as long as we consistently live at head :+1: . An `-rc` version does have its merits working on a final product/app, but i don't think there's that much of a point for us since we're working on the library level. There's no way to do additional 'rc'-like testing anyway without someone actually trying to use it in a final product.

closed 2024-03-05 by mgovers