PowerGridModel/power-grid-model #501: [BUG] In validation function, it still requires power sigma for power sensor

TonyXiang8787, 2024-02-19:
In the validation function for state estimation, `power_sigma` is still required. However, we either need `power_sigma`, or `p_sigma` and `q_sigma`.

The validation function should be changed.

mgovers, 2024-02-20:
I remember that, by design, we decided that `power_sigma` always needs to be provided. At least, it is reflected in the docs in that way: https://power-grid-model.readthedocs.io/en/stable/user_manual/components.html#generic-power-sensor . Therefore, this is not a bug report, but a new feature request.

See also the discussion in #409 . that way, there's always a fallback value.

TonyXiang8787, 2024-02-20:
@mgovers now after my second thought it actually makes sense to require either `power_sigma`, or `p_sigma` and `q_sigma`. Instead of always require `power_sigma`. This is confusing for the user.

closed 2024-03-01 by Jerry-Jinfeng-Guo