PowerGridModel/power-grid-model #508: [FEATURE] Check for necessary condition for observability

mgovers, 2024-02-29:
## Describe the feature request

Add a check that implements a necessary condition for observability listed in the [state estimation documentation](https://power-grid-model.readthedocs.io/en/stable/user_manual/calculations.html#state-estimation).

## Rationale

This is a first step towards #84 that implements the design + a simple check.

## Out of scope

Sufficient condition for observability in state estimation. The necessary and sufficient condition for observability is covered in #84 . Instead, as mentioned in #rationale , this is a first step that is already able to design and implement an initial check, which can then be extended on in #84

## Design

* [ ] Create a separate module with checks for observability.
* [ ] When running any state estimation solver
  * [ ] First run the observability checks.
  * [ ] If the observability check fails, raise an error
* [ ] TBD: raise `SparseMatrixError` vs creating a new `NotObservableError`

mgovers, 2024-02-29:
> TBD: throw SparseMatrixError vs creating a new NotObservableError

@TonyXiang8787 do you have any thoughts on this?

TonyXiang8787, 2024-02-29:
> > TBD: throw SparseMatrixError vs creating a new NotObservableError
> 
> @TonyXiang8787 do you have any thoughts on this?
> 
> AFAICT, they are two separate things and it should be a new `NotObservableError`.
> 
> Rationale:
> 
> * A non-sufficiently filled Ybus or other calculation parameters may still cause `SparseMatrixError`s. E.g.: Powerflow calculations may throw them as well.
> * Not all unobservable cases result in sparse matrix errors

If it fails the necessary condition check, it should throw `NotObservableError`. The places where `SparseMatrixError` is thrown, we keep them as is.

mgovers, 2024-03-01:
> If all connecting edges are already enabled, quit the function. (undetermined situation)

so this is the case where Marc's work will provide a necessary and sufficient condition, right?

TonyXiang8787, 2024-03-01:
> > If all connecting edges are already enabled, quit the function. (undetermined situation)
> 
> so this is the case where Marc's work will provide a necessary and sufficient condition, right?

That's one, the other one is when graph has cycles (with enabled edges).

mgovers, 2024-03-01:
I'm thinking whether we can approach this in a 2-phase way (with incremental improvements).

The main body of this PR can be implemented with just this simple check without any graph search required:

[image]

and then after that's merged, a more complex check with the above logic can be implemented.

TonyXiang8787, 2024-03-01:
@mgovers After some thinking, I concluded that the graph search will not provide additional information. If the first three condition pass, the graph search will either conclude that all vertices are reachable, which is a **sufficient condition**, or result in undermined quit. It will never issue a failure on check of reachable vertices. I will put that part check into an invalid section.

mgovers, 2024-03-18:
Closed by #523

closed 2024-03-18 by mgovers