PowerGridModel/power-grid-model #529: [FEATURE] Static library wrapper around main model + math solver for speeding up tests

mgovers, 2024-03-13:
Relates to https://github.com/PowerGridModel/power-grid-model/issues/486

A major factor in the compilation is the fact that all 3 test files of the main model + the validator + the benchmark need to recompile all templates individually. Making a static library wrapper would reduce this to only 1 compilation.

The way to do that is the following:

* [ ] Add a separate library to the tests directory: `tests/cpp_test_wrappers/`
  * [ ] Add a `CMakeLists.txt` to expose a `STATIC` library `cpp_test_wrappers`
  * [ ] Add a header `tests/cpp_test_wrappers/include/cpp_test_wrappers/main_model_wrapper.hpp`
    * [ ] declare class:
      ```cpp
      class MainModelWrapper {
        private:
          class Impl;

        public:
          // all public methods of MainModel

        private:
          std::unique_ptr<Impl> impl_;
      }
      ```
  * [ ] Add a cpp file `tests/cpp_wrappers/src/main_model_wrapper.cpp`
    * [ ] declare class impl:
          ```cpp
          class MainModelWrapper::Impl : public MainModel {
            public:
              using MainModel::MainModelImpl;
          };
          ```
    * [ ] implement all public methods of MainModelWrapper by forwarding the call to `MainModel`
    * [ ] Add all static checks from `test_main_model_static.cpp` to this file
* [ ] Modify `test_main_model_*.cpp`, the `cpp_validation` tests and the benchmark to use this library instead of the static `MainModel`
* [ ] Optionally: do the same for the MathModel
  * [ ] Split the MathModel tests into separate files

TonyXiang8787, 2024-03-15:
I wonder if we are creating static library anyway, why don't we just make `power_grid_model` core library itself a static library. We try to compile the header files in several compilation unit.

mgovers, 2024-03-15:
I've been thinking about that as well. Let's discuss offline

mgovers, 2024-07-11:
As discussed: we will solve this in a different way

closed 2024-07-11 by mgovers