PowerGridModel/power-grid-model #530: [BUG] Cannot release due to broken github runner image

mgovers, 2024-03-14:
**Describe the bug**

CI fails on Linux C++ builds

**Input Data Validity**

Did you check the validity of input data using the helper function? 

yes. Builds and runs just fine locally

**Data Dump**

https://github.com/PowerGridModel/power-grid-model/actions/runs/8273113443/job/22646069378

**To Reproduce**

Run CI on `main`; see that it fails

**Expected behavior**

CI passes on main; new version is released

**Screenshots**

N/A

**Additional context**

Most likely related to https://github.com/actions/runner-images/issues/9491

TonyXiang8787, 2024-03-14:
@mgovers can we revert to gcc-12 at this moment?

mgovers, 2024-03-14:
> @mgovers can we revert to gcc-12 at this moment?

that's not the issue. As can be seen in the github issue, the problem happens on both Clang and GCC. It seems to be related to the github runner images being broken - not because of anything by us.

it happens more often now than earlier this week because the new runner images are now fully rolled-out, Earlier this week, a rerun was sufficient if we were lucky enough to being scheduled on a runner that was not migrated to the latest image yet.

TonyXiang8787, 2024-03-15:
@mgovers can we try this work around? So we are not stucked.

https://github.com/actions/runner-images/issues/9491#issuecomment-1989718917

mgovers, 2024-03-15:
> @mgovers can we try this work around? So we are not stucked.
> 
> [actions/runner-images#9491 (comment)](https://github.com/actions/runner-images/issues/9491#issuecomment-1989718917)

I discussed yesterday with @petersalemink95 that we would not be doing that but given the amount of open PRs at this point, I do think it's worth at least checking whether that fix solves the issue

TonyXiang8787, 2024-03-15:
> I discussed yesterday with @petersalemink95 that we would not be doing that but given the amount of open PRs at this point, I do think it's worth at least checking whether that fix solves the issue

Otherwise we are stucked with so many PR's until GitHub fixes the issue. Maybe still re-consider the work around.

mgovers, 2024-03-15:
In the meantime, I already did a test on an old version of the PGM before any CI issues started on Ubuntu (`v1.7.6`, so we know that MacOS will fail) in https://github.com/PowerGridModel/power-grid-model/actions/runs/8292801374/job/22694713820 . If that Ubuntu build passes then further investigation is definitely required; if not - I will create another run with the proposed solution from the github runners issue

TonyXiang8787, 2024-03-15:
Another workaround is to disable `clang` in the cpp test at this moment. The bug only affects llvm as I read.

mgovers, 2024-03-15:
> In the meantime, I already did a test on an old version of the PGM before any CI issues started on Ubuntu (`v1.7.6`, so we know that MacOS will fail) in [...]

yup, also resulted in segfaults, as expected. Will investigate the workaround.

> Another workaround is to disable `clang` in the cpp test at this moment. The bug only affects llvm as I read.

GCC + address sanitizer may also be affected, but since clang compiles faster, that's the one failing. I'll create a draft PR with the workaround and just kickoff CI and see what happens

closed 2024-03-15 by TonyXiang8787