PowerGridModel/power-grid-model #582: [BUG] Serialization should fail with a readable error when required memory cannot be allocated

nitbharambe, 2024-04-30:
**Describe the bug**

Serialization fails with an difficult to interpret error message. Likely cause as per the msgpack error meassage is that required memory cannot be allocated.

**To Reproduce**

A big output dataset was attempted to serialize to file using json and msgpack serializer.

**Expected behavior**

Some understandable error message like `PowerGridSerializationError` would make it clear.

**Screenshots**

json_serialize_to_file gives out following:
[image]

same data with msgpack export it gives out `bad allocation` message.

TonyXiang8787, 2024-05-01:
@nitbharambe, what is the use case to export large dataset to `json`?

nitbharambe, 2024-05-01:
The dataset is only relatively huge. Intention was to store intermediary results of an analysis. (Although there can be ways around it.)

[image]

> @nitbharambe, what is the use case to export large dataset to `json`?

mgovers, 2024-05-01:
a couple observations:

* [ ] a good starting point might be to determine how many mega/gigabytes the data to be allocated was (e.g.: less or greater than 4GB due to page size). Depending on the specific parameters provided to the serializer, this may be calculated from the above dataset size
* [ ] the difference in behaviour most likely comes from the fact that the default result value is `char const*{}` which is `NULL`. For the JSON serializer, this is interpreted as null terminated string for JSON (illegal for null, hence the stacktrace) and to fixed-size string for `msgpack`
* [ ] it might be a good idea to make a separate `PGM_serialize_to_file` functionality that directly outputs the result data to the filestream instead of first to python. If the `std::bad_alloc` came from e.g. the page size, then directly outputting to file might actually fix the issue.

PR #635 (2024-06-07): [Fix] handle nullptr on serialization error
Closes #582 

Fix unintelligible error message when serializing to JSON fails.

The behavior to msgpack has not changed because "bad allocation" really is the only valid error message: we don't know what or where it goes wrong, but there was a memory allocation that was bad.

## Current `main` (before bugfix)

Behavior depends on serialization format

### serialize to msgpack

Throw `PowerGridSerializationError` with the message `bad allocation`

[image]

### serialize to JSON

Throw `AttributeError` with the message `'NoneType' object has no attribu [... cut]
closed 2024-06-07 by Jerry-Jinfeng-Guo