PowerGridModel/power-grid-model #595: [FEATURE] Output correct automatic tap even when line drop compensation is large

mgovers, 2024-05-13:
During implementation of #588 , it was discovered that, when line drop compensation is large, there are cases where the assumption that the voltage on the controlled side decreases with increasing tap position doesn't hold anymore. The current optimization logic in the automatic tap optimizer strongly depends on that assumption.

The edge case arises when there is a phase angle between U and I, and $\underline{I} * \underline{Y}_{compensation}$ is relatively large compared to U.

**NOTE:** The outcome of this discussion may affect #594

## Example toy model

As an example, consider a toy model of a load with constant power, i.e., $\underline{U} * \underline{I}$ remains constant.

Consider U and I to be in phase (both real numbers) $U_{comp} = |U + (R + iZ) I|$ and everything is p.u., such that $U=1/t$ and $P = U I = 1$ , so that $I = t$ . The full equation now becomes $U_{comp}(t) = |1/t + (R + iZ) t|$

For ease of illustration, now consider the (extreme) case where $Z=0$ and $R=0.1$, so that $U_{comp}(t) = |1/t + 0.1 t| = |(1 + 0.1 t²)/t|$. For integer $t$, $U_{comp}$ is decreasing for $0 < t <= 3$ (as per the assumption), but it is increasing for $t >= 4$.

This function is plotted in https://www.wolframalpha.com/input?i=abs%281%2Ft+%2B+%280.1%29+*+t%29

TonyXiang8787, 2024-05-13:
The `max_iter` of the outer loop (per rank) can be automatically determined by the difference between `tap_min` and `tap_max` of all transformers in this rank.

mgovers, 2024-05-16:
Conclusion of discussion:

* Edge cases like this are rare in real grid scenarios.
* They can theoretically happen for generators but we don't expect them to.
* It is OK to just let the algorithm diverge in these cases, and call it undefined behavior (UB).
* No changes need to happen in the short term.
* The following check may be implemented as an input validation function in the long term.

[image]