PowerGridModel/power-grid-model #662: [FEATURE] Automatic tap changer in step-up transformer support

Jerry-Jinfeng-Guo, 2024-07-04:
What need to happen:
- [ ] Figure out the meeth
- [ ] Control side at the node with higher rated voltage 
- [ ] Tap side and the control side at the same side

Jerry-Jinfeng-Guo, 2024-08-15:
Most changes will happen in `power_grid_model_c/power_grid_model/include/power_grid_model/optimizer/tap_position_optimizer.hpp`

What needs to happen:
* [ ] validation to check for unsupported scenarios when it comes to enabled tap controllers:
 * [ ] more than one enabled tap controller on the same transformer
 * [ ] an enabled tap controller is coupled with a not-energized transformer 
 * [ ] an enabled tap controller is coupled with a two/three winding transformer with disconnected tap side or control side
 * [ ] an enabled tap controller is coupled with a two winding transformer which are not fully connected
 * [ ] an enabled tap controller is coupled with a three winding transformer with two sides disconnected
 * [ ] an enabled tap controller is coupled with a transformer which is not reachable by transformer ranking graph, see logic below
* [ ] un-regulated transformers given weight of `0` instead of `1`:
      in lines 119-120 and 157-158, from
       ```c++
            add_to_edge(state, edges, edge_props, from_node, to_node, {unregulated_idx, 1});
            add_to_edge(state, edges, edge_props, to_node, from_node, {unregulated_idx, 1});
       ```
       to
       ```c++
            add_to_edge(state, edges, edge_props, from_node, to_node, {unregulated_idx, 0);
            add_to_edge(state, edges, edge_props, to_node, from_node, {unregulated_idx, 0});
       ```
* [ ] logic determining direction of transformer tap/control sides needs update:
  * [ ] two winding transformer in lines 151-153
  * [ ] three winding transformer in lines 107-109 
* [ ] after assigning ranks of all regulated transformers, one round of check, between line 274 and 276
  * [ ] two winding transformers: for every enabled such transformer, rank at the control side (node connected to it) should be 1 greater than the other side, throw exception if not the case.
  * [ ] three winding transformers: 
    * [ ] control side == tap side: rank at the control side (node connected to it) should be 1 greater than any one of the two secondary sides, throw exception if not the case.
    * [ ] control side != tap side: rank at the control side (node connected to it) should be 1 greater than the secondary side with tap, throw exception if not the case.
* [ ] Adjust control logic 
  * [ ] lines 941-950, linear search would have to include logic to go the opposite way if control side == tap side
  * [ ] lines 634-667, binary search would have to add new state `control_tap_same_side` in the binary logic

mgovers, 2024-08-16:
line numbers can change. can you add permalinks? (you can create them using the GitHub viewer)

closed 2025-03-14 by Jerry-Jinfeng-Guo