PowerGridModel/power-grid-model #669: [BUG] Validation fails for a configuration without three winding transformers

wfjvdham, 2024-07-09:
**Describe the bug**

I get an KeyError: <ComponentType.three_winding_transformer: 'three_winding_transformer'> when validating a configuration that does not have any `three_winding_transformers`. 

**To Reproduce**

```python
del input_data["three_winding_transformer"]
assert_valid_input_data(input_data)
```

**Expected behavior**

I think it should be possible to validate an input_data object without three_winding_transformer?

**Additional context**

Can be solved by changing:

https://github.com/PowerGridModel/power-grid-model/blob/b1903af981f2fad4897793f8000f3c9a56fbb2d8/src/power_grid_model/validation/rules.py#L919-L923

into:

```python
for ref_component, ref_field in tap_side_fields:
    if data.get("ref_component", False):
      indices = get_indexer(data[ref_component]["id"], data[component][regulated_object_field], default_value=-1)
      found = indices != -1
      ref_comp_values = data[ref_component][ref_field][indices[found]]
      invalid[found] = np.logical_or(invalid[found], values[found] == ref_comp_values)
```

mgovers, 2024-07-09:
Hi Wim,

Thank you for reporting.

I will investigate and will get back to you as soon as I have an update.

mgovers, 2024-07-09:
Hi @wfjvdham ,

After a very quick investigation, I verified that this indeed a bug and your proposal is the right way to go to fix the issue (with some minor adjustments). Thank you for your clear and elaborate input.

We will merge this ASAP.

wfjvdham, 2024-07-09:
Cool! Thanks for the quick response

mgovers, 2024-07-09:
FYI @wfjvdham , the fix in #670 has been merged, but it may take up til an hour from now until CI finishes and `v1.9.4` containing the fix is released.
* If you are using PyPI, the release will be available using `pip install` or equivalent immediately after the release.
* If you are using `conda-forge`, then it may take up to 18 hours before it becomes available if everything goes correctly.

mgovers, 2024-07-09:
Closely related follow-up found in #671 . A fix will be available in a couple hours (most likely `v1.9.5`)

commit 4dd45b04 (2024-07-09): Merge pull request #670 from PowerGridModel/bugfix/fix-669-regulator-without-other-trafo-type

fix #669 transformer tap regulator without other transformer type in data
commit a8b2ceb3 (2024-07-09): fix #669 transformer tap regulator without other transformer type in data
PR #671 (2024-07-09): fix also support empty trafo list in validation
Closes #669 

Relates to #670 .

This is much less severe but still really annoying.
closed 2024-07-09 by mgovers