PowerGridModel/power-grid-model #720: [FEATURE] support asymmetric line

leovsch, 2024-09-12:
# Background
Currently the `line` component only supports symmetric components as input. Namely, `r1` `x1` `r0` `x0`. 

## Feature request
Opendss also supports the full specification of Rmatrix and Xmatrix see [the opendss manual](https://opendss.epri.com/Line.html) section 
'Matrix Impedance Definition Properties'. We would like PGM to also support this by having an new component called `asym_line` so that we can reproduce the same result as OpenDSS.

petersalemink95, 2024-09-13:
Hi @leovsch,

Thank you for your request; this would be a nice addition to Power Grid Model.
I heard from @TonyXiang8787 that you're interested in picking-up the feature yourself. 
That would be awesome.
I'll add the steps that need to be taken to add a new component to Power Grid Model below.
Know that the maintainers are there to support you, if you need help.

You can either request help through the issue or get our contacts directly through @TonyXiang8787.

- [ ] Check if the available input/update/output data suffices (check `power-grid-model/power_grid_model_c/power_grid_model/include/power_grid_model/auxiliary/` `input.hpp` / `update.hpp` / `output.hpp` or in the documentation directly)
- [ ] If not, add the new data format to `code_generation/data/attribute_classes/` `input.json` / `update.json` / `output.json`  + run `code_generation/code_gen.py`
- [ ] Create a new component in a new `power-grid-model/power_grid_model_c/power_grid_model/include/power_grid_model/component/[component].hpp` file that at least inherits from Base, but in this case `GenericBranch` should inherit from `Branch`
- [ ] If necessary: add new enums or exceptions
- [ ] Create the necessary unit tests in `power-grid-model/tests/cpp_unit_tests/test_[component].cpp`
- [ ] Add the `test_[component].cpp` to `power-grid-model/tests/cpp_unit_tests/CMakeLists.txt`
- [ ] Add component to `power_grid_model_c/power_grid_model/include/power_grid_model/all_components.hpp`
- [ ] Not necessary for this component (If necessary update `main_core/topology.hpp` / `input.hpp` / `output.hpp` / `update.hpp`)
- [ ] Add component to `code_generation/data/dataset_class_maps/dataset_definitions.json` + re-run `code_generation/code_gen.py`
- [ ] Add validation test cases to `tests/data`
- [ ] Update input/update data validator for the new component: `src/power_grid_model/validation/validation.py` + add corresponding tests

Note: the order is recommended, but not necessary

leovsch, 2024-10-03:
Hi @petersalemink95,

Thanks for the implementation steps. I've started with them but I'm running into a build error.
Whenever I run `pip install -e .[dev]` as described in the [documentation](https://power-grid-model.readthedocs.io/en/stable/advanced_documentation/build-guide.html).

I get the following error:

```
C:\Users\20180029\repos\power-grid-model>pip install -e .[dev]
Obtaining file:///C:/Users/20180029/repos/power-grid-model
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... error
  error: subprocess-exited-with-error

  × Getting requirements to build editable did not run successfully.
  │ exit code: 1
  ╰─> [39 lines of output]
      Build wheel from C:\Users\20180029\repos\power-grid-model\setup.py
      Download package: eigen, version: 3.4.0 ...
      Download package: boost, version: 1.86.0 ...
      Traceback (most recent call last):
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 132, in get_requires_for_build_editable
          return hook(config_settings)
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\setuptools\build_meta.py", line 464, in get_requires_for_build_editable
          return self.get_requires_for_build_wheel(config_settings)
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\setuptools\build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\setuptools\build_meta.py", line 302, in _get_build_requires
          self.run_setup()
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\setuptools\build_meta.py", line 318, in run_setup
          exec(code, locals())
        File "<string>", line 255, in <module>
        File "<string>", line 251, in prepare_pkg
        File "<string>", line 162, in generate_build_ext
        File "<string>", line 51, in get_pre_installed_header_include
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\pybuild_header_dependency\header_resolver.py", line 30, in __init__
          self._resolve_pkgs()
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\pybuild_header_dependency\header_resolver.py", line 61, in _resolve_pkgs
          self.downloaders[name].download(version, self.pkg_path)
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\pybuild_header_dependency\custom_sources\boost.py", line 37, in download
          self.unpack_files(
        File "C:\Users\20180029\AppData\Local\Temp\pip-build-env-i54_bkiq\overlay\Lib\site-packages\pybuild_header_dependency\package_downloader.py", line 38, in unpack_files
          tf.extractall(tmp_dir)
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\tarfile.py", line 2059, in extractall
          self.extract(tarinfo, path, set_attrs=not tarinfo.isdir(),
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\tarfile.py", line 2100, in extract
          self._extract_member(tarinfo, os.path.join(path, tarinfo.name),
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\tarfile.py", line 2173, in _extract_member
          self.makefile(tarinfo, targetpath)
        File "C:\Users\20180029\AppData\Local\Programs\Python\Python310\lib\tarfile.py", line 2214, in makefile
          with bltn_open(targetpath, "wb") as target:
      FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\20180029\\AppData\\Local\\Temp\\pybuild-header-dependency\\pkgs\\tmp\\boost_1_86_0\\libs\\numeric\\odeint\\doc\\html\\doxygen\\odeint_reference\\classboost_1_1numeric_1_1odeint_1_1checked__stepper_3_01_controlled_stepper_00_01_checker_00_01controlled__stepper__tag_01_4.html'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build editable did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

[notice] A new release of pip is available: 24.0 -> 24.2
[notice] To update, run: python.exe -m pip install --upgrade pip
``` 
I am using windows 11 and have visual studio with c++ development modules installed. Are you familiar with this error and do you know how to fix it?

Thanks!

TonyXiang8787, 2024-10-03:
Hi @leovsch,

You are running into a classical Windows long path problem. Have a look at the Windows manual below:

https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry

It is also mentioned in the Windows example:
https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry

closed 2025-04-29 by nitbharambe