PowerGridModel/power-grid-model #723: [FEATURE] Placement of the from and to in variables

wfjvdham, 2024-09-17:
**Describe the feature request**

I think it would be nice to be consistent with where to put the from and to. Right now it is `to_node`, `to_status`, but also `winding_to`, `i_to`

TonyXiang8787, 2024-09-17:
Hi @wfjvdham,

Thanks for the feedback. It is indeed important to keep the name consistent. However, changing existing attribute names will be a huge breaking change for the users and we cannot justify that.

In the future attributes, we will keep the names more consistent.

mgovers, 2024-09-17:
One thing we can do is to have attribute types similarly to how we now have component types, in which' case the new one is an alias for the old one so that it's not a breaking change.

petersalemink95, 2024-09-20:
> One thing we can do is to have attribute types similarly to how we now have component types, in which' case the new one is an alias for the old one so that it's not a breaking change.

In that case you will have to have documentation for both. That would become messy / less readable. It is a fair request from @wfjvdham, but unfortunately we cannot change this anymore.

TonyXiang8787, 2024-09-20:
Observing the discussion I hereby close this issue. We will pay more attention to make the future attribute names as consistence as possible.

closed 2024-09-20 by TonyXiang8787