PowerGridModel/power-grid-model #729: [FEATURE] Provide a generic branch

petersalemink95, 2024-09-23:
Following the discussion in #642 it was concluded that the implementation of a generic branch could be an addition to Power Grid Model.

We can make a new component `generic_branch`, which model branches as a Pi model; i.e. the user can specify:
- k: transformer ratio (float)
- theta: angle shift
- r_series
- x_series
- g_shunt
- b_shunt

##steps to follow
In order to add a new component to Power Grid Model the following steps need to be taken:

- [ ] Check if the available input/update/output data suffices (check `power-grid-model/power_grid_model_c/power_grid_model/include/power_grid_model/auxiliary/` `input.hpp` / `update.hpp` / `output.hpp` or in the documentation directly)
- [ ] If not, add the new data format to `code_generation/data/attribute_classes/` `input.json` / `update.json` / `output.json`  + run `code_generation/code_gen.py`
- [ ] Create a new component in a new `power-grid-model/power_grid_model_c/power_grid_model/include/power_grid_model/component/[component].hpp` file that at least inherits from Base, but in this case `GenericBranch` should inherit from `Branch`
- [ ] If necessary: add new enums or exceptions
- [ ] Create the necessary unit tests in `power-grid-model/tests/cpp_unit_tests/test_[component].cpp`
- [ ] Add the `test_[component].cpp` to `power-grid-model/tests/cpp_unit_tests/CMakeLists.txt`
- [ ] Add component to `power_grid_model_c/power_grid_model/include/power_grid_model/all_components.hpp`
- [ ] Not necessary for this component (If necessary update `main_core/topology.hpp` / `input.hpp` / `output.hpp` / `update.hpp`)
- [ ] Add component to `code_generation/data/dataset_class_maps/dataset_definitions.json` + re-run `code_generation/code_gen.py`
- [ ] Add validation test cases to `tests/data`
- [ ] Update input/update data validator for the new component: `src/power_grid_model/validation/validation.py` + add corresponding tests

Note: the order is recommended, but not necessary

TonyXiang8787, 2024-09-23:
The user needs to be able to provide the attributes of `r`, `x`, `g`, `b`, `k`, `theta` for positive, negative and zero sequence. So the attribute list is actually 3 times as described.

TonyXiang8787, 2024-09-24:
Hi @sudo-ac, @petersalemink95,

Now after re-think about the issue maybe the way that we define attribute as `r/x_series` and `g/b_shunt` is not very logical. The grounding impedance of the transformer will be appended in one of the `pi` model. Making shunt part of the `pi` model not symmetric between `from` and `to` side. In the mathematical model the branch is described as:

$$
\begin{bmatrix}
I_f \\
I_t
\end{bmatrix} =
\begin{bmatrix}
Y_{ff} & Y_{ft} \\
Y_{tf} & Y_{tt}
\end{bmatrix}
\begin{bmatrix}
U_f \\
U_t
\end{bmatrix}
$$

The above formula applies three time for positive, negative and zero sequence. Using for complex parameters $Y_{ff}$, $Y_{ft}$, $Y_{tf}$, $Y_{tt}$ we can take the tap ratio, phase shift, grounding, and everything into account. 

Maybe we need to let user directly define those instead of  `r/x_series` and `g/b_shunt`, which is a half-way solution.

TonyXiang8787, 2024-09-24:
> Hi Tony (@petersalemink95),
> 
> this means you prefer to define the input parameters in this way:
> 
> r0_series / r1_series / r2_series x0_series / x1_series / x2_series g0_shunt / g1_shunt / g2_shunt b0_shunt / b1_shunt / b2_shunt
> 
> 0 = zero sysytem 1 = positiv system 2 = negative system

Hi @sudo-ac,

Actually, not after my rethinking. The current way (in the issue) we define series impedance and shunt admittance does not allow us to give different values for the two legs of the `pi` model, which is needed for the transformer modelling. My new idea would be asking user to provide attributes which are closer to the internal mathematical model, namely `yff_1`, `yft_1`, `ytf_1`, and `ytt_1`. Same for `_2` and `_0`. But it is a lot of attributes. Note in the interface we do not support complex numbers, so it eventually becomes `gff_1` and `bff_1`, etc.

petersalemink95, 2024-09-24:
@TonyXiang8787 The idea of `generic_branch` was to make the life of the end user easier, by not having to do conversions of the data themselves if R/X/G/B are present. I think letting the end user calculate all ff/ft/tf/tt parameters does not fit the original purpose of the issue anymore...

What do you think @sudo-ac?

TonyXiang8787, 2024-09-24:
But now we are running into issue that the transformer has not the same values in the two legs of the `pi` model.

We can still use the R/X/G/B model, then we have to use like `g_shunt__from_1` and `g_shunt_to_1`.

petersalemink95, 2024-09-24:
@sudo-ac We're also thinking hard about the best solution.
The asymmetric calculation is a bit harder than expected here.

To include you in our thinking:
-For transformers there is a deviation between the `from` and `to` side of `b0` and `g0` (zero sequence). Therefore the original proposed idea is not valid.
- Adding `r_grounding_from`, `x_grounding_from`, `r_grounding_to`, `x_rounding_to` as you proposed can be a solution for transformers, but for transformers only; making it not a `generic_branch` anymore.
- Adding `y_ff`, y_ft`, `y_tf` and `y_tt` , as @TonyXiang8787 proposed, can be a solution, but this would put more work at the user for asymmetric calculations

At this moment, do you just want to do symmetric calculations? If that is the case we can also implement symmetric calculations first and have some more time to think about the best asymmetric implementation: throw a NotImplementedError for now. 

We have a brainstorm with the team tomorrow about this topic. If you have any ideas on how you would like to have this implemented from a user perspective, please let us know!

petersalemink95, 2024-09-24:
Thanks @sudo-ac. I'll come back to you after the brainstorm tomorrow!

petersalemink95, 2024-09-26:
Hi @sudo-ac thanks for thinking along!

We've had a brainstorm with the team and concluded that the asymmetric part is quite difficult and needs some more work.
Since you're only interested in doing symmetric calculations at the moment you can go ahead by only providing the symmetric calculation (and positive sequence r,x,g,b). For the asymmetric calculation you can throw a `NotImplementedError`. I hope this idea works for you as well!

I will make a new issue for the asymmetric calculation, including some of the options we discussed.

petersalemink95, 2024-09-27:
Hi @sudo-ac,

That is great to hear! Once the PR is made we'll do the review.
Regarding #739: there is no decision yet on how to implement asymmetric calculations. For us it would be fine to go ahead by just merging the symmetrical generic_branch for now; if that would suit your needs.

TonyXiang8787, 2024-11-21:
This issue will be closed. The asymmetric implementation is further defined in #739

closed 2024-11-21 by TonyXiang8787