PowerGridModel/power-grid-model #733: [FEATURE] Ideal source handling

mgovers, 2024-09-24:
## Describe the feature request

An ideal source is represented by having infinite short circuit power (`sk`). However, the current implementation of the power-grid-model will raise a validation error when `sk == inf`:
[image]

This ticket is about supporting ideal sources for power flow and state estimation calculation types. (And undefined behavior for short circuit calculations.)

## Background

When `sk` is extremely large but not infinite (e.g. `sk == 10e50`), the power flow and state estimation calculation types are already numerically stable, cfr. #458 . Unfortunately, this value is arbitrarily set. We currently require our users to provide such an arbitrarily large value when they desire an ideal source. Even more so, within the PGM, intermediate calculations may cause an arbitrarily large value to become infinite, e.g.:

* `10e200 * 10e200 = 10e400 > std::numeric_limits<double>::max() (~10e306)` => `10e200 * 10e200` will result in `inf`

A single such `inf` value in the internal calculations will result in `NaN` values in the output, i.e. nonsense output.

## Desired solution

* [ ] In the C++ implementation of the Source component type in https://github.com/PowerGridModel/power-grid-model/blob/main/power_grid_model_c/power_grid_model/include/power_grid_model/component/source.hpp :
  * [ ] Create a `static constexpr double ideal_source_sk = 10e50;` constant.
    * [ ] Note that this is arbitrarily chosen (mention this in a comment)
    * [ ] This value is so large that it does not affect numerical stability for practical use cases, while still being small enough to never cause infinities and NaN values in the calculations. (please verify with simple examples)
  * [ ] In the constructor of the Source component type, set `sk` to the minimum of the user-specified `sk` and `ideal_source_sk`.
    * [ ] Note that this is only correct for power flow and state estimation
    * [ ] For short circuit calculations, the `sk` heavily influences the calculation result, and we consider this to be undefined behavior (UB; see below)
* [ ] Update the validation cases
  * [ ] Support `sk == inf` for power flow and state estimation
  * [ ] Do not support `sk == inf` and `sk > 10e50` for short circuit calculations due to deliberately UB
* [ ] In the documentation on the Source component type : 
  * [ ] On the input attributes: https://github.com/PowerGridModel/power-grid-model/blob/main/docs/user_manual/components.md?plain=1#L388
    * [ ] Mention that `sk == inf` represents an ideal source
  * [ ] On the electrical model in https://github.com/PowerGridModel/power-grid-model/blob/main/docs/user_manual/components.md?plain=1#L392-L416 :
    * [ ] Mention that the `sk` is capped at `10e50`
    * [ ] Mention that unrealistically large `sk` (`sk > 10e50`) results in UB for short circuit calculations, because the `sk` heavily affects the short circuit calculation results.

VITA2aishu, 2026-09-14:
Hi, I’m interested in working on this issue. I’d be happy to implement the ideal source handling along with the required validation tests and documentation updates. Please let me know if I can take this up.

Jerry-Jinfeng-Guo, 2026-09-17:
Hi @VITA2aishu , thank you for your interest in this feature. It is indeed available. Please let us know if you need any help in the process. As a reminder, please make sure that your commits are signed and certified.

closed 2026-10-22 by figueroa1395