PowerGridModel/power-grid-model #739: [FEATURE] asymmetric implementation of generic_branch

petersalemink95, 2024-09-26:
For the discussions about `generic_branch` see #729.

The asymmetric implementation is more difficult and needs more thinking. Some options could be:

- Provide r1, x1, b1, g1, r2, x2, b2, g2 for positive and negative sequence and y_ff, y_ft, y_tf, y_tt for zero sequence
- Provide r_series / x_series and g/b_shunt_from/to
- Provide r1, x1, b1, g1, r2, x2, b2, g2, r0, x0, b0, g0, so provide the PI model for all sequences: this way only YnYn transformers can be calculated in asym calculation

mgovers, 2024-09-26:
maybe `GenericLine` and `GenericTransformerYNyn`, ...?

TonyXiang8787, 2024-09-27:
Hi @sudo-ac, thanks for the suggestion. 

If we define so many `GenericBranch*`, I wonder what generic we do have here. We really need to think hard about how to implement asymmetric generic branch. @petersalemink95, please also add this option in the list of choices to make.

For now, is it sufficient for you @sudo-ac to first implement the symmetric `GenericBranch` for your own use-cases? If yes, we are looking forward to seeing a PR about that.

petersalemink95, 2024-09-27:
@TonyXiang8787 added it as an option to this issue.

@sudo-ac Thanks for the suggestion! As @TonyXiang8787  mentioned we might need to think a bit longer about the best solution. Would just implementing symmetric calculations for now suit your need? Your proposal would still be valid if we decide to implement that in a later stage if we implement only the symmetric calculations now.

petersalemink95, 2024-09-30:
Thanks for all the input @sudo-ac!
We're looking forward to reviewing the symmetric PR! (But enjoy your holiday first ;) )

We will include your latest suggestions in further discussions about the asymmetric generic branch.

petersalemink95, 2024-10-11:
Hi @sudo-ac,

Thanks for working out your idea!
I would like to propose to wait with further investigation of asymmetric generic branch until a user presents itself. 
That way we can design the correct implementation including the perspective of the actual user.

Let's keep this issue open, so further discussions can take place here.