PowerGridModel/power-grid-model #773: [FEATURE] Bump github actions to Ubuntu 24.04

figueroa1395, 2024-10-09:
**Describe the feature request**

On October 3th, 2024, `ubuntu-release` will be [fully migrated](https://github.com/actions/runner-images/issues/10636) to `ubuntu-24.04` for github actions. This feature request is about upgrading along to `ubuntu-24.04`. For reference [here](https://github.com/actions/runner-images/blob/main/images/ubuntu/Ubuntu2404-Readme.md) are the list packages, tools, etc. that come with this image and [here](https://github.com/actions/runner-images/issues/9848) is the announcement of this image and its implications.

**Steps**

This is the list of things that need to be done to do the full migration:

- [ ] Upgrade to `ubuntu-24.04` (or `ubuntu-release` after October 30th) for github actions.
- [ ] Upgrade `clang`,  `clang-tidy` and `clang-format` to version 18.1.3. This involves also updating `llvm` to version 18.
- [ ] Upgrade `gcc` and `g++` to version 13 (14 is experimental in `ubuntu-24-04`).
- [ ] Upgrade `python` to 3.12 in the github actions.
- [ ] Possibly upgrade to `cmake` 3.30 (Not necessary, but now supported in ubuntu-24.04).
- [ ] Update `pre-commit` accordingly.
- [ ] Update build-guide (and other) documentation accordingly.

**Experiment**

An experiment was run on #770 to see what would the impact be to do the migration, and the main conclussions are:

- The validate-citations action [initially broke](https://github.com/PowerGridModel/power-grid-model/actions/runs/11236794248/job/31237704644?pr=770). A quick fix is implemented in https://github.com/PowerGridModel/power-grid-model/pull/770/commits/7f8ee76382f4335f8cfe4366e42a937f1c569f9b.
- If `clang-format` is upgraded, the [check-code-quality](https://github.com/PowerGridModel/power-grid-model/actions/runs/11236794230/job/31237705042?pr=770) action breaks and many files need to be re-formatted again (also pre-commit needs to align with the versions here)
- If  `clang` (but not `clang-tidy`) is upgraded, the [clang-tidy](https://github.com/PowerGridModel/power-grid-model/actions/runs/11250947913/job/31280919270?pr=770) action breaks horribly. If both `clang` and `clang-tidy` are upgraded, [many fixes](https://github.com/PowerGridModel/power-grid-model/actions/runs/11251673483/job/31283225777?pr=770) need to happen accross the core, as the clang-tidy action breaks. These probably need to be evaluated in a per-case scenario.

The above seem to be the only problems with the migration (without considering the `cmake`upgrade which may be a follow up).

**Takeout**

The hard deadline to take action is October 30th, 2024, so we probably need to either address the above issues or pin github actions to `ubuntu-22.04`.

mgovers, 2025-02-26:
First attempt in https://github.com/PowerGridModel/power-grid-model/pull/882

figueroa1395, 2025-03-12:
@mgovers We can close this issue as completed in favor of having merged #882, correct? Before doing so, should I make a small PR to replace `ubuntu-24.04` to `ubuntu-latest`? Only remaining is to update the runner on `read the docs`; I can quickly take a look if 24.04 is supported already there, and estimate the effort and value of doing so.

mgovers, 2025-03-12:
> [@mgovers](https://github.com/mgovers) We can close this issue as completed in favor of having merged [#882](https://github.com/PowerGridModel/power-grid-model/pull/882), correct? Before doing so, [...]

Yes, looks like everything worked. I actually think we can already close this. Readthedocs is a nice to have but less important (in the sense that: as long as it still builds, we're good to go). You can have a look indeed

closed 2025-03-12 by mgovers