PowerGridModel/power-grid-model #786: [FEATURE] Columnar data creation helper function

nitbharambe, 2024-10-15:
A `initialize_column(data_type, component, shape, attributes, empty=False)` helper function which gives out columnar data ie. dict of arrays.

## Motivation:

We have `initialize_array` helper function to make users not bother about the dtype. A same one for columnar data.

If the user has to create a columnar data now, they would have to do :
`{k: v for k,v in initialize_array(dataset_type, component_type, shape).items()}` 
or 
`{k: np.empty(shape=...,  dtype=power_grid_meta_data[dataset_type][component_type][k]) for k in attributes}` 
or some similar combination

Also in first way, user creates a huge array; subset of which is then converted to dictionary.

mgovers, 2024-10-24:
`initialize_array(data_type, component, shape, empty=False, *, attributes=None)`:

petersalemink95, 2024-11-06:
> decision: use `initialize_array(data_type, component, shape, empty=False, *, attributes=None)` with additional kwarg

To be clear: `attributes=None` should refer to the row based initialization, that already is present. For columnar people should specify a "filter" in `attributes=`, similar as in the data filter

Jerry-Jinfeng-Guo, 2024-11-21:
@nitbharambe Is this still open?

closed 2024-11-21 by TonyXiang8787