PowerGridModel/power-grid-model #796: [BUG] Simple update with batch datasets in Mac yields error

figueroa1395, 2024-10-21:
**Describe the bug**

As found in the investigation #790, when doing a permanent simple update `model.update()` with a batch dataset (batch size > 1), there is a bug that only happens on Mac and on certain architectures that made the [feedstock CI crash](https://github.com/conda-forge/power-grid-model-feedstock/pull/272) (not our own).

**To Reproduce**

A detailed explanation with investigation can be found in #790, but to simply reproduce, make a branch from main and revert https://github.com/PowerGridModel/power-grid-model/pull/790/commits/795b7e9c1c57d50d270b55ab12266a2e8f217f14, then build and run using the same [configuration ](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=1054810&view=results)on Mac as the failed azure pipelines.

**Expected behavior**

Feedstock CI should not break as this feature is [properly tested](https://github.com/PowerGridModel/power-grid-model/pull/790) and [passes in our CI](https://github.com/PowerGridModel/power-grid-model/pull/778).

**Additional context**

After #790, it was found that permanent updates with batch datasets were not properly tested, and the necesary tests were added in the core in the same PR, showing that the core indeeds support it. Additionally, the root of this issue maybe related or similar to https://github.com/PowerGridModel/power-grid-model/pull/769.

mgovers, 2024-10-21:
FYI: renamed `Simple update` to `Permanent update` because it carries the meaning better. This is not a big problem, just a minor thing. "Simple update" may appear more severe to a user than it actually is.

figueroa1395, 2025-01-20:
> [@figueroa1395](https://github.com/figueroa1395) is this still an issue?

I'll try locally (now that I can) and report back later. I ad forgotten about this, thanks for the reminder.

figueroa1395, 2025-01-22:
> [@figueroa1395](https://github.com/figueroa1395) is this still an issue?

@mgovers I tried to reproduce locally on the Mac M3 by reverting https://github.com/PowerGridModel/power-grid-model/commit/795b7e9c1c57d50d270b55ab12266a2e8f217f14 without success. Since I can't reproduce locally, it still only fails on the [feedstock CI for Mac](https://github.com/conda-forge/power-grid-model-feedstock/pull/272) (Runners perhaps are of an older architecture?).

However, since the functionality is actually properly tested in the [core](https://github.com/PowerGridModel/power-grid-model/pull/790/commits/75337c6279122ee17cafb9280e60e7484c1a9481), I would argue that we close this issue for now and only come back to it if a user has a problem. What do you think?

mgovers, 2025-01-22:
I'm OK with that. @petersalemink95 @TonyXiang8787 do you agree?

TonyXiang8787, 2025-01-22:
Let's close this.

closed 2025-01-22 by TonyXiang8787