PowerGridModel/power-grid-model #800: [BUG] Potential memory leak

figueroa1395, 2024-10-21:
**Describe the bug**

While working on https://github.com/PowerGridModel/power-grid-model/pull/795, it was found that there is a memory leak somewhere in the core, which is not catched by the CI and it was only detected by the Address Sanitizer locally in my PC. Precisely, the bug is related with the test cases `"Update error"` and `"Batch calculation error"` in the test file `test_api_model.cpp`. Furthermore, the tests pass sucessfully (value wise), but Address Sanitizer still catches a leak.

More specifically, both leaks are caused exactly by the line `input_dataset.add_attribute_buffer("line", "id", line_id.data());` in `test_api_model.cpp`, and, weirdly enough, if this line is removed, there is no leak, but the tests still pass and there is no hard crash as it should be expected (since we would not be providing update ids and https://github.com/PowerGridModel/power-grid-model/pull/771 is not ready yet).

**To Reproduce**

I run `test_api_model.cpp` in VS Code, in Windows 11, via WSL2 with Ubuntu 22.04, using gcc 11.4.0 in debug mode with address sanitizer activated.

**Screenshots**

A simple screenshot showing part of the issue is shown for a bit of context, more information can be provided later if needed:
[image]

**Additional context**

After working on a minimal reproducible example, by removing not-needed stuff from the core and reducing the test scope to only where this issues are present, the following was concluded:

- The error is there for row, columnar and mixed based data; so this was (likely) not introduced with recent features regarding columnar data.
- If instead of 2 lines, we only introduce one, the leak is reduced from 32 bytes to 16, as expected.
- A quick (and dirty) "fix" was found: if in the definition of `AllComponents` in `all_components.hpp`, the order of `Line` and `Source` is exchanged (`Source` appears on the list before `Line`), the memory leak is "removed". In addition, when debugging, many `assert`s checks in `Branch` are not reached when they should to prevent/flag this. In spite of this finding, this is not a true solution, as the underlying cause is still unknown and needs further investigation.

figueroa1395, 2024-10-24:
After further investigation, it was found that the memory leak can only be reproduced specifically in WSL with Ubuntu 22.04 with gcc 11. Ubuntu 22.04 with gcc 12, and Ubuntu 24.04 with gcc 12 and 13 were tested without any related problems. Hence, it is very likely that is is an Address Sanitizer false positive (bug) or a gcc 11 compiler issue (less likely). Furthermore, the decision was to not address this right now and wait until we drop support for manylinux_2_24 (approximately in mid 2025) to deprecate gcc 11.

Jerry-Jinfeng-Guo, 2024-10-24:
Thanks for the handling!

closed 2024-10-24 by Jerry-Jinfeng-Guo