PowerGridModel/power-grid-model #984: [FEATURE] PGM Meet-up 2025-05-21 Hackaton: Validation cases from publicly available datasets

nitbharambe, 2025-05-02:
# Validation cases from publicly available datasets

The modelling of various components and the algorithms for various calculations within power-grid-model are validated against established power system analysis tools and hand calculations. This ensures theoretical accuracy and reliability of the developed features. Power-grid-model (PGM) includes over 80 validation cases that are utilized to validate every incremental development of PGM. While all features are validated with minimal component examples, various practical or research focused larger grids can also be tested.
Since power-flow analysis results cannot be directly obtained from the inputs, these validation cases enhance confidence in the results produced by the PGM tool. The validation cases consist of a test input, an expected output, and a configuration file for the calculation. This task involves:
- Modelling publicly available  grids in PGM using available IEEE datasets of PES test feeders.
- Converting the solutions to the PGM output format, which serves as the expected output for the calculation.
- Creating the validation case and contributing to the PGM repository.
- Verifying if the results match.

## Steps
- [ ] Install `power-grid-model` along with development dependencies in Python.
- [ ] (Optional) Create a script to generate data directly from the table provided below.
- [ ] Download the case data from IEEE PES Test Feeders.
- [ ] Develop an input data model for PGM using the test cases, which should include topology, parameters, loads, and generation information from the feeder data.
- [ ] Specify any mutations to the input data in the update data section, such as load profiles for timeseries simulations.
- [ ] Create the expected output for the previously generated input/update data by referring to the solutions section.
  - You may choose to validate only specific results. For instance, if power results are adequately validated, voltage results may be omitted at your discretion (or vice-versa).
- [ ] Define the configuration parameters in `params.json`.
- [ ] Execute the validation tests.

## optional steps for creating script to create data from tables:
- [ ] Read the tables.
- [ ] Use `initialize_array` to fill the input PGM array. Check [Components — power-grid-model documentation](https://power-grid-model.readthedocs.io/en/stable/user_manual/components.html) for details on all attributes.
- [ ] Fill the update PGM array.
- [ ] (Optional) Run PGM data validator: [`assert_valid_input_data`](https://power-grid-model.readthedocs.io/en/stable/user_manual/data-validator.html#utilities) to identify data errors in input.
- [ ] Fill the output PGM array.
- [ ] Export to a validation test case (TODO).

VITA2aishu, 2026-09-05:
Hi, I'd like to contribute to this issue. I'm interested in adding one publicly available IEEE test feeder as a power-flow validation case. Since the issue covers multiple possible datasets, is there a particular small feeder/test case you would recommend as a good first contribution? I'm happy to start with one case and follow the existing validation dataset structure.

Jerry-Jinfeng-Guo, 2026-09-07:
Hi @VITA2aishu , thank you for your interest in contributing to our test set! We would welcome any public datasets. For the IEEE networks, perhaps you could start with the 14-bus network, as it is still manageable in size. We appreciate single test as much as batch test.

furqan463, 2026-10-15:
Reference to discussion in https://github.com/PowerGridModel/power-grid-model-io/pull/499
Following challenges have been identified for creating validation cases for IEEE 4-node Test Feeder

1. `power-grid-model` does not support `delta` connected load. In order to model loads in `delta`, additional virtual node and `Dy` transformer is required. IEEE solutions does not contain the expected results of these additional nodes.
2. For nodes connected to delta winding, IEEE published solution for voltage is `L-L`, while `power-grid-model` validation data is expected in `L-N`. Since zero-sequence voltage is absent from `L-L` voltage, `L-N` voltage cannot be calculated from published results.
3. Line current magnitude and angle are published. current magnitude can be directly compared, but `power-grid-model` does not produce line current angle. Reference to point (2) above, for delta connected side, `L-N` voltage is not available in published results therefore we cannot calculate powers (P,Q) using current and voltage results.

For above reasons, a validation case cannot be constructed from `power-grid-model`. However, we can create a customised validation script in python for validation of results against published results.

Moreover, Open-winding transformer is currently not supported by `power-grid-model`, so those variants cannot be converted.

furqan463, 2026-10-26:
> 15\. Following challenges have been identified for creating validation cases for IEEE 4-node Test Feeder

Further, for remaining IEEE PES Test feeders following challenges are faced:

1. distributed loads are not supported in `power-grid-model`. One workaround could be to introduce a virtual node in the centre of the section and add load on that node.
2. Asym voltage regulators (with taps/ voltage regulation for individual phases) is not supported in `power-grid-model`, therefore, all the test feeders other than IEEE 4-node cannot be converted to `power-grid-model`.