PowerGridModel/power-grid-model #990: [BUG] PGM dependency issue in pandapower since PGM v1.10.107

pawellytaev, 2025-05-08:
**Describe the bug**
Dear PGM developers,

In pandapower we are currently facing issues due to the dependency on PGM. The tests for runpp_pgm are currently failing in pandapower, see a few examples here: 
https://github.com/e2nIEE/pandapower/pull/2601
https://github.com/e2nIEE/pandapower/pull/2391

I managed to reproduce this issue locally, by using PGM v1.10.108. When rolling back versions locally, to v1.10.106, my tests are successful locally - therefore I suspect this issue coming from the 1.10.107 release.

Could you please give us a hint where this issue might come from? By scrolling through the recent PGM commit history, I couldn't get a clue unfortunately :-(

Thanks in advance.

Pawel

**Input Data Validity**

N/A

**Data Dump**

N/A

**To Reproduce**

- create a fresh conda environment
- install the latest version of pandapower from develop with pip install -e .["all"] ("all" in order to get all pandapower dependencies, including PGM)
- run pytest on pandapower/test/loadflow/test_runpp_pgm.py
- --> tests are failing
- pip install power_grid_model==1.10.106
- run pytest again
- --> tests are successful

**Expected behavior**

Successful tests in pandapower with latest PGM release. Alternatively, we will fix the dependency to PGM <= 1.10.106

**Screenshots**

failed test from https://github.com/e2nIEE/pandapower/actions/runs/14907111629/job/41871990973?pr=2601 

[image]

pawellytaev, 2025-05-08:
@panos-xenos @vogt31337

mgovers, 2025-05-08:
Dear @pawellytaev , thank you for reporting. This is indeed a bug and was also reported in https://github.com/PowerGridModel/power-grid-model-io/issues/297 . A bugfix was proposed in #988 and released in [v1.10.109](https://github.com/PowerGridModel/power-grid-model/releases/tag/v1.10.109). Can you please confirm whether it is resolved?

Announcements regarding this issue will follow (c.c. @nitbharambe @Jerry-Jinfeng-Guo @petersalemink95 )

pawellytaev, 2025-05-08:
Thanks for the fast response!
Unfortunately the tests are still failing, see here a screenshot of my local test:

[image]

and the failed jobs here after re-running them a few minutes ago:
https://github.com/e2nIEE/pandapower/actions/runs/14908893199/job/41883696010?pr=2391

mgovers, 2025-05-08:
> Thanks for the fast response! Unfortunately the tests are still failing, see here a screenshot of my local test:
> 
> [image]
> 
> and the failed jobs here after re-running them a few minutes ago: https://github.com/e2nIEE/pandapower/actions/runs/14908893199/job/41883696010?pr=2391

Apologies and thank you for checking. The problem was fixed via #991 and released in [v1.10.110](https://github.com/PowerGridModel/power-grid-model/releases/tag/v1.10.110). Can you please double-check whether the issue was resolved correctly this time?

pawellytaev, 2025-05-09:
v1.10.110 resolves this issue, thanks :)

Closing this issue then

closed 2025-05-09 by pawellytaev