PowerGridModel/power-grid-model #992: [FEATURE] PGM Meet-up 2025-05-21 hackathon: Create enumerations for all attributes used in PGM

nitbharambe, 2025-05-09:
# Description

Power grid model data in python API has multiple dataset types, components and each of which contain multiple attributes (see [Input dataset example]( https://power-grid-model.readthedocs.io/en/stable/examples/Power%20Flow%20Example.html#input-dataset)). We can access each required components and attributes directly via strings. This is not the a best practice as it causes multiple issues, some of which are mentioned in this issue [#613]( https://github.com/PowerGridModel/power-grid-model/issues/613). Towards this end, PGM has implemented DatasetType and ComponentType enums in [#648]( https://github.com/PowerGridModel/power-grid-model/pull/648). The enums for attributes however are yet to be implemented.

This hackathon task is about adding all attribute related enums in PGM for the Python API. After design discussions within PGM team, it is decided to create a single enum with attributes present in it. With new support, the number of attributes keep on increasing, hence, this enum has to be generated automatically via code generator to simplify further development.

# Setup

Ensure the following tools are installed. Download them from their official websites:
- git
- Python
- VS Code or other IDE.

Follow these steps to set up your environment for development in the Python API of PGM. Adjust them as needed based on your system configuration:

- Clone the git repository of PGM.
```{sh}
git clone https://github.com/PowerGridModel/power-grid-model.git
```
- Navigate to the repository directory, create a virtual environment, and activate it:
```{sh}
cd path/to/power-grid-model
python -m venv .venv
.venv\Scripts\activate
```
- (TODO: Discuss making C++ core available.)
- Install power-grid-model in development mode.
```{sh}
pip install -e .[dev]
```
- Verify the setup by running all tests:
```
pytest
```

# Task
Following steps are adapterd from the instructions specified in [#613]( https://github.com/PowerGridModel/power-grid-model/issues/613). Refer the issue for details.
- [ ] Set up code generator to give out the enum. 
- [ ] Run the code generator
- [ ] Expose the new enum in the __init__.py
- [ ] Update the documentation to reflect the changes.
- [ ] Update examples to use the new enum.
- [ ] Update PGM [workshop](https://github.com/PowerGridModel/power-grid-model-workshop) notebooks.
- [ ] Update attributes in all places at code in PGM.
- [ ] Update tests at PGM
- [ ] Update PGM-IO with new enums

petersalemink95, 2025-05-14:
Original issue in #949

petersalemink95, 2025-08-11:
Closing, since the meet-up is done. Original issue still remains

closed 2025-08-11 by petersalemink95