PowerGridModel/power-grid-model #1100: [FEATURE] Add utility function to reorganize scenarios according to the performance guide

mgovers, 2025-08-28:
**Describe the problem**

Different use cases require different sets of calculations. Since the PGM treats scenarios in an isolated fashion, the core is agnostic to the order of calculations.

However, some caching may happen in the PGM core to speed up the calculations. But whether and what can be cached depends on the order and contents of the scenarios in the update data, as well as amount of threads used during the calculation.

Guidelines to speed up the calculation are documented in https://power-grid-model.readthedocs.io/en/stable/user_manual/performance-guide.html#batch-calculations .

For many use cases, the order of scenarios does not really matter to the user (e.g., when doing N-1 simulations). Even when the order does matter (e.g., when doing time series calculations), only the final order matters.

**Improvement proposal**

* [ ] Add a function `reorder_scenarios_for_performance(*, BatchDataset update_data) -> tuple[BatchDataset, list[int]]` to the [power_grid_model.utils module](https://github.com/PowerGridModel/power-grid-model/blob/main/src/power_grid_model/utils.py) that:
  * [ ] creates an ordering `reordering: list[int]` of the scenarios (following the conventions of [`np.argsort`](https://numpy.org/devdocs/reference/generated/numpy.argsort.html#numpy.argsort)) in the update data according to the performance guidelines, that is:
    * [ ] it keeps scenarios with the same topology (e.g., same switch statuses) together
    * [ ] within the same topology, it keeps scenarios with the same electrical parameters (e.g., branch parameters, tap position, ...) together
  * [ ] creates a copy (!!!) of the update data with the reordered order of scenarios, such that it is equivalent to
    ```py
    for reordered_idx, scenario_idx in enumerate(reordering):
        np.testing.assert_array_equal(
            get_dataset_scenario(reordered_update_data, reordered_idx),
            get_dataset_scenario(update_data, scenario_idx)
        )
    # for row-based dense batch data, this is equivalent to:
    for component in update_data:
        np.testing.assert_array_equal(
            reordered_update_data[component][:],
            update_data[component][reordering]
        )
    ```
  * [ ] computes the `inverse_ordering: list[int]` of the scenarios, such that
    ```py
    for scenario_idx, reordered_idx in enumerate(inverse_ordering):
        np.testing.assert_array_equal(
            get_dataset_scenario(reordered_update_data, reordered_idx),
            get_dataset_scenario(update_data, scenario_idx),
        )
    # for row-based dense batch data, this is equivalent to:
    for component in update_data:
        np.testing.assert_array_equal(
            reordered_update_data[component][:],
            update_data[component][inverse_ordering],
        )
    ```
  * [ ] returns both the `reordered_update_data` and the `inverse_ordering`
  * [ ] works for both row-based, columnar data, dense and sparse data, as well as a mixture of the two
  * [ ] has a clear docstring
* [ ] Add tests for row-based, columnar, dense, sparse and mixed data.

nitbharambe, 2026-08-31:
Hi @appleweiping , nice that you are picking this up. 

1. Good call. I imagine the resultant ordering output would be different in both situations. However there is no target date for that feature yet. I think it would be best if we keep both issues (This and #1103) separate. (Unless you or someone else wish to pick that issue up as well, then situation changes! :D). We should also highlight in the docs / docstring that output of this function may change as per the latest improvement / features.

2. No issue in adding input_data keyword arg in my opinion. And we can set it optional and conservatively group assuming the input_value != nan and input_value != update_value. 

3. Maybe we can start this implementation with hardcoded params. Lets scope this out. An ideal solution can be related to #1519. 

4. Yes what you mention should be the behaviour. The dense inverse example is incorrect. Ill modify it such that inverse_ordering applies on the reordered_update_data such that we can construct original input data.

5. Sounds good.

TonyXiang8787, 2026-08-31:
@mgovers @nitbharambe  so your proposal is only a Python convenienc function instead of C-API core feature?

nitbharambe, 2026-08-31:
@TonyXiang8787 Indeed. We did not receive any specific request from C API users as of yet. Hence we decided to start with a python utility initially as it is easier to implement.

nitbharambe, 2026-08-31:
@appleweiping Since its been an year this issue was posted, please hold on the implementation for some time while we discuss requirements further within the users / maintainers.

appleweiping, 2026-08-31:
Thanks for the clarification. I’ll hold off on implementation while the users and maintainers refine the requirements. I’ve noted the current direction—an initial Python utility, separate from #1103, with optional input data, focused hardcoded classification, and the confirmed ordering/inverse-ordering contract—but I won’t treat it as final or start coding until you confirm the scope is ready. Please tag me when the discussion has converged and I’ll be happy to continue.

nitbharambe, 2026-09-10:
@appleweiping 
Based on the points raised by you and other maintainers, I would close this issue. Reasons being:
As mentioned, there is a continuous effort on maintenance corresponding to new attributes and with #1103.
It was openend primarily because we did not have a method to easily cross chain of batches. That itself does not solve the issue of reorganization but combined with reading the performance docs it is an easy task for user with this usecase. Setting this case aside, there is less value that can be brought in.

If there is interest from community, please drop a post here and we shall reopen this.

closed 2026-09-10 by nitbharambe